u-root/u-root #58: Need command completion in shell

rminnich, 2016-06-07:
We don't want to add too many features, but people want command completion. We don't want to recreate bash, however.

ryukinix, 2016-10-12:
This can be done directly on the rush program core?

rminnich, 2016-10-12:
yeah but first I'd suggest you look around and see what other go-based shells are out there, maybe there's something we want more. Although the simplicity of rush has its appeals. I do NOT want a bash-in-go -- the scripting language is Go, not something else :-)

rafaelcn, 2016-10-13:
I'd suggest to use an array to store the last `n` commands typed and when a specific key is pressed it just gets the `n-1` command, what do you think @rminnich and @ryukinix?

rminnich, 2016-10-13:
why not just store all the commands every typed? Let's go further. Why not store commands in a map, rather than a linear list? Let's not limit ourselves to how things are done now. But I'm glad you are thinking about these things.

rafaelcn, 2016-10-14:
That might slow the shell at some time - no evidence to support my assertion -, I guess we could put a very high limit instead and also why a map? Is any advantage in that?

rminnich, 2016-10-14:
You need to measure. The map would ensure there is only one copy of each command you type in the history.

ryukinix, 2016-10-14:
Yes. The map is a hash tree, so we don't have multiplies equal keys, right @rminnich ?

GanShun, 2016-10-14:
With the map, do we care about the order of commands? with a list you have the history in linear order. with the map do we want to keep track of the order in which the commands were executed?

rminnich, 2016-10-14:
gan, good point, but ryukinix it is a hash tree.

I think Gan's point about order is a good one. I just don't want to lock us in to old thinking based on shells written in C.

rjoleary, 2017-10-21:
Closing because the elvish shell fills this need.

closed 2017-10-21 by rjoleary