u-root/u-root #60: We need a more capable startup daemon in Go

rminnich, 2016-06-07:
We have a /init already. We could describe dependencies in JSON files, have it compute the dependencies, and execute commands.

stephenmw, 2017-11-26:
Even better would be to use https://github.com/google/skylark. Compatibility with sdnotify would also be cool.

hugelgupf, 2017-11-26:
Skylark would be interesting for this. Though I think bazel doesn't actually use that implementation of it? But it's probably good enough anyway :)

bensallen, 2017-12-07:
# U-root Init Proposal

## Service Files

- 3 ways to define services: 
 
1. Go package, compiled directly into init via import.
2. Go plugin, compiled during init run-time and later via a call to reload service file. Allows for a dynamic way of adding and removing Go based services from a running init.
3. Service configuration file. Limited to executing commands (eg. Systemd unit file)

### Example service

```
package foo

import (
	"fmt"

	"github.com/u-root/u-root/init/pkg/service"
)

//foo is a unit todo bar
type foo service.Unit

//New returns a service.Unit of foo
func New() service.Servicer {
	return foo{
		Type:            service.Simple,
		Before:          []string{},
		After:           []string{},
		Requires:        []string{},
		EnvironmentFile: "",
		PIDFile:         "",
	}
}

func (f foo) Start() error {
	fmt.Println("Hello world")

	return nil
}

func (f foo) Stop() error {
	fmt.Println("Goodbye world")

	return nil
}

func (f foo) Restart() error {
	f.Stop()
	f.Start()

	return nil
}

func (f foo) Reload() error {
	f.Restart()

	return nil
}
```

- A service must implement a Servicer:

```
type Servicer interface {
	Start() error
	Stop() error
	Reload() error
	Restart() error
}
```

### Notes

- Each service can implement a flagset, so for example one could run `init foo reload` only if reload is implemented.
- On start or plugin reload we'll need build a dependency graph of services.
- A generator for adding imports and \<service\>.New() calls could be used.
- Should create a generator for a skeleton Go service package and plugin source file

### Notes on plugins

- Compiled file size of plugins is large, so we should compile source into plugin at run time and on change of service Go file.
- Allows for dynamic inclusion of service go files
- Plugins not supported on all platforms :-(, looks like just Linux currently.

bensallen, 2017-12-12:
Summarizing some discussion in Slack:

1. Services are processes
2. Monitoring/control is via some ReST-ful like thing
3. There is a service for all services that follows these rules

bensallen, 2018-02-01:
Testing out building an init with Hashicorp's go-plugin here: https://github.com/bensallen/service-plugin

bensallen, 2018-02-12:
# U-root Init Proposal (Updated)

## Components

## Init

- Init spawns a manager process.
- Init should attempt recovery of manager if the manager dies.

### Manager

- Launches servicer processes
- Computes directed graph of dependancies to determine start order and services that can be launched in parallel
- Manager exposes Boot() "StartAll", Shutdown() "StopAll", Recover() RPC for init
- Exposes RPC Servicer interface for CLI tools, eg. Status(), Start(), Stop(), Restart(), Reload(), and Unit() of individual services
- Reports a map (eg. state) of running servicers back to init on a regular basis for the goal of being able to recover manager crashing.

### Servicer / go-plugins / service binary

- Implements Servicer interface, eg Status(), Start(), Stop(), Restart(), Reload(), and Unit()
- Exposes Servicer interface via RPC using go-plugin
- Defines a service.Unit, overridable via config file /etc/service.d/foo.conf. 
- Include a generator for skeleton service binary
- Real services based on existing init.go in u-root that need to be implemented
    - Filesystems, parse fstab(?)
    - buildbin
    - Background Build
    - Goenvs
    - Getty / login

### Interface

- A service must implement a Servicer:

```
type Servicer interface {
	Start() error
	Stop() error
	Reload() error
	Restart() error
        Status() state.Value
        Unit() Unit
}
```

### Example service

```
package main

import (
	"os"
	"time"

	"github.com/hashicorp/go-hclog"
	"github.com/hashicorp/go-plugin"
	"github.com/u-root/service-plugin/pkg/service"
	"github.com/u-root/service-plugin/pkg/service/onfail"
	"github.com/u-root/service-plugin/pkg/service/state"
)

// foo is a bar
type foo service.Unit

var logger = hclog.New(&hclog.LoggerOptions{
	Level:      hclog.Trace,
	Output:     os.Stderr,
	JSONFormat: true,
})

//New returns a service.Unit of foo
func New() service.Servicer {

	return &foo{
		Name:            "foo",
		Description:     "foo does all of the bar",
		Type:            service.Simple,
		OnFail:          onfail.Restart,
		Before:          []string{},
		After:           []string{},
		Requires:        []string{},
		Enabled:         true,
		EnvironmentFile: "/etc/service.d/foo.conf",
		State:           state.Unknown,
	}
}

func (f *foo) Unit() service.Unit {
	return service.Unit(*f)
}

func (f *foo) Start() error {
	time.Sleep(5 * time.Second)
	logger.Debug("Hello world")
	f.State = state.Starting

	return nil
}

func (f *foo) Stop() error {
	logger.Debug("Goodbye world")
	f.State = state.Stopping

	return nil
}

func (f *foo) Restart() error {
	f.Stop()
	f.Start()

	return nil
}

func (f *foo) Reload() error {
	f.Restart()

	return nil
}

func (f *foo) Status() state.Value {
	return f.State

}

func main() {

	foo := New()

	// pluginMap is the map of plugins we can dispense.
	var pluginMap = map[string]plugin.Plugin{
		"foo": &service.Wrapper{Impl: foo},
	}

	plugin.Serve(&plugin.ServeConfig{
		HandshakeConfig: service.HandshakeConfig,
		Plugins:         pluginMap,
		Logger:          logger,
	})
}
```

hugelgupf, 2018-02-12:
I like it, I think (haven't thought through it all, tbh). 

One thing about existing init - as a migration path, we don't need to immediately convert all of current init to this. In fact, in bb mode that's not possible yet. (But it's also likely that you don't want any of this in bb mode.) So as a starting point, this init system could be uinit while we make some structural improvements to init in the different modes so it can be swapped out easily. I imagine some of the source and bb mode specific stuff will have to stick around for a little while anyway. See #615.

rjoleary, 2021-06-07:
This has already been done in Nichrome as "Service of Services". See "sos", "timesos" and "upspinsos" here: https://github.com/u-root/NiChrome/tree/master/cmds. You can build these into your u-root initramfs.

There are instructions for making your own service and example here: https://github.com/u-root/NiChrome/tree/master/examples/sos

I'd say this is a fairly capable startup daemon, so this issue can be closed.

closed 2021-06-07 by rjoleary