u-root/u-root #61: RUN_ME_FIRST error

funespy, 2016-06-08:
Hi, I was trying to run the RUN_ME_FIRST file but I got theses errors.

[image]

This is the output
`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   318    0     0    0     0      0      0 --:--:--  0:00:06 --:--:--     0
100  4682  100  4682    0     0    450      0  0:00:10  0:00:10 --:--:--  2428
2016/06/07 23:41:45 Using /usr/local/go from the environment as the GOROOT
2016/06/07 23:42:23 Go get failed: err exit status 1, output 
[103 111 32 105 110 115 116 97 108 108 32 114 117 110 116 105 109 101 47 105 110 116 101 114 110 97 108 47 115 121 115 58 32 111 112 101 110 32 47 117 115 114 47 108 111 99 97 108 47 103 111 47 112 107 103 47 108 105 110 117 120 95 97 109 100 54 52 47 114 117 110 116 105 109 101 47 105 110 116 101 114 110 97 108 47 115 121 115 46 97 58 32 112 101 114 109 105 115 115 105 111 110 32 100 101 110 105 101 100 10]
2016/06/07 23:44:33 Go get failed: err exit status 1, output 
[103 111 32 105 110 115 116 97 108 108 32 114 117 110 116 105 109 101 47 105 110 116 101 114 110 97 108 47 115 121 115 58 32 111 112 101 110 32 47 117 115 114 47 108 111 99 97 108 47 103 111 47 112 107 103 47 108 105 110 117 120 95 97 109 100 54 52 47 114 117 110 116 105 109 101 47 105 110 116 101 114 110 97 108 47 115 121 115 46 97 58 32 112 101 114 109 105 115 115 105 111 110 32 100 101 110 105 101 100 10]
`

Can you guide me to solve this?

Thanks.

rafaelcn, 2016-06-09:
Hm, this is very weird. @rminnich could you please give us a light?

rminnich, 2016-06-10:
This is because our RUN_ME_FIRST is quite out of date! I am working to fix it. Thanks for the report!

rminnich, 2016-06-10:
Can you both take a look at at my PR?

rafaelcn, 2016-06-10:
Looks good to me Ron, though I don't really tested it out.

rafaelcn, 2016-10-03:
Isn't this error caused by shouting the error with the `%v` instead of `%s`? Just have a look at this [line](https://github.com/u-root/u-root/blob/master/scripts/getimports.go#L207).

rafaelcn, 2017-03-08:
Looks like this was already addressed, therefore, is going to be closed.

closed 2017-03-08 by rafaelcn