u-root/u-root #63: Code-review: what about barkeep?

ryukinix, 2016-07-12:
Hey @rminnich, what do you think about this one? I think a code review system like gerrit is very nice, but [BarKeep](http://getbarkeep.org/) seems is more pragmatic.

rafaelcn, 2016-08-27:
The tool seems very intuitive, far simple than gerrit! But as we discussed in bringing this - the project - to Github there's no point in moving back to a code review again I think.

ryukinix, 2016-08-27:
I think the productivity of group was better with a code review system.

ryukinix, 2016-09-21:
I think github code review system is pretty good. Maybe we can disable the pullapprove and use just the codereview system of github. @rafaelcn, @rminnich

rafaelcn, 2016-09-21:
I agree with this as well. Github review system has improved, actually, it now has one.

closed 2016-10-04 by ryukinix