u-root/u-root #82: Enable website: why not?

ryukinix, 2016-10-04:
We need just this to enable our website and get visible.

`mv website docs`

And activate gh-pages by:

[image]

I think we can create a more friendly documentation on README.md too.

rminnich, 2016-10-04:
YEAH!
LGTM

ryukinix, 2016-10-12:
@rminnich, you can create a pull-request for this? I wanna work on docs with some friends. We think really great if the README.md and website has a clear and usefull documentation about the project (how does, why this, what the deal and so on).

closed 2016-10-12 by ryukinix

ryukinix, 2016-10-14:
We need setup yet the GitHub Pages as described in the picture of this issue. I can't do that, just you @rminnich. Would be cool a CNAME like u-root.tk? (I can take for free)

rminnich, 2016-10-14:
What permissions do you need to do this? I will give them to you.

ryukinix, 2016-10-14:
I need be owner from organization to change that type of things, the settings of repository. You can go to here https://github.com/orgs/u-root/people and change this. After that I can setup this.

rminnich, 2016-10-14:
Done.

PR #87 (2016-10-12): website: make it live.
closes #82
commit 9771ab60 (2016-10-12): website: make it live.

closes #82