u-root/u-root #93: Naive attempt to test u-root, problem with cpio

ryukinix, 2016-10-16:
```
2016/10/16 12:44:57 exit status 1
2016/10/16 12:44:57 External cpio is done
2016/10/16 12:44:57 Done all cpio operations
2016/10/16 12:44:57 Creating initramf file
2016/10/16 12:44:57 Run &{/usr/bin/sudo [sudo cpio -H newc -o -A -F /tmp/initramfs.linux_386.cpio] [] /tmp/u-root247321542 0x1862c748 0x185260e0 0x185260e8 [] <nil> <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>} @ /tmp/u-root247321542
2016/10/16 12:44:57 Finished sending file list for initramfs cpio
sudo: cpio: command not found
2016/10/16 12:44:57 exit status 1
2016/10/16 12:44:57 cpio for initramfs is done
2016/10/16 12:44:57 Run &{/usr/bin/sudo [sudo cpio -i] [] /tmp/u-root247321542 0x1862c870 0x185260e0 0x185260e8 [] <nil> <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>} @ /tmp/u-root247321542
sudo: cpio: command not found
2016/10/16 12:44:57 exit status 1
exit status 1
```

I got this after execute `go run scripts/ramfs.go -test -d`. Maybe we need put `cpio` as dependency on docs?

rminnich, 2016-10-17:
yes we do!
Thanks!

rjoleary, 2017-02-04:
Ever since @rminnich implemented u-root's cpio command, it might be possible for `ramfs.go` to use u-root's version of cpio, alleviating the need to install an additional dependency.

```
$ go install github.com/u-root/u-root/cmds/cpio
$ ls "$GOPATH/bin"
cpio  golint
$ cpio --help
Usage of cpio:
  -v	Debug prints
```

PR #404 (2017-11-05): Factor out initramfs code into package; share in bb and scripts/ramfs.go
This is the first step in unifying the ramfs and bb scripts. Eventually, the "common" package will be renamed to something else, but I'll figure that out as I go along.]

Will fix #93
closed 2017-11-05 by hugelgupf