u-root/u-root #94: How abou a verbose mode on scripts/ramfs.go?

ryukinix, 2016-10-16:
We, at least my machine, the ramfs.go now really take a good time to get into a u-root shell. The debug mode has a lot of information, but is really useful for debug. How about create a `-v` option to put on stdout few information about what u-root are doing on ramfs.go? Is really scary don't know what he's doing on the normal mode. I think we don't need do too much, like a progressbar of compiling. Just print the sections of procedures to get into the ramfs would be good... searching libs,local variables, compiling source, preparing chroot and etc. A minimal verbose mode. What you do think, guys?

rafaelcn, 2016-10-16:
Ron already did that, look at the u-root slack and also into ramfs.go code, it has a -v option and also a ludicrous (or something like that) one.

rminnich, 2016-10-17:
Can you see if -v works for you?
Thanks!

ryukinix, 2016-10-17:
```
u-root git/master  26s
❯ go run scripts/ramfs.go -test -v
flag provided but not defined: -v
Usage of /tmp/go-build327305449/command-line-arguments/_obj/exe/ramfs:
  -cpio string
    	An initial cpio image to build on
  -d	Debugging
  -removedir
    	remove the directory when done -- cleared if test fails (default true)
  -test
    	test the directory by chrooting to it
  -tmpdir string
    	tmpdir to use instead of ioutil.TempDir
  -useinit
    	If there is an existing init, don't replace it
exit status 2
```

rafaelcn, 2016-10-17:
Well, that's weird. Did you saw the code inside ramfs?

ryukinix, 2016-10-17:
Not yet, because I just was walking around and checking the executions I propose that issue. Where is the problem?

rafaelcn, 2016-10-17:
Ok, the `-v` flag should be there as Ron commit #86 shows that.

ryukinix, 2016-10-18:
That question is: anybody try use that on head of master?

closed 2017-10-30 by hugelgupf