u-root/u-root #116: Some commands with good code coverage show 0% coverage under `go test --cover`

rjoleary, 2017-01-25:
At the moment, some of our tests with coverage are reported as 0% (ex: sort_test.go and uniq_test.go). The problem results from executing a process within the test via "go run" or "go build". The coverage tools aren't smart enough to descend into the child processes to count their coverage. Additionally, the child processes are not built with the proper instrumentation.

While a bit dated, the post here gives some useful background on the issue: https://blog.golang.org/cover

Things I've tried so far:

- Executing the test within the test to create a subprocess. This technique is used by some tests in the go standard library. Unfortunately, the coverage array is not copied to the parent process.
- Use the Go cover tool when building the child process's binary. Unfortunately, the coverage array is not copied to the parent process.

The only way I see of fixing this is using a function call rather than spawning a child process.

Relevant research:

- http://stackoverflow.com/questions/40615641/testing-os-exit-scenarios-in-go-with-coverage-information-coveralls-io-goverall
- https://talks.golang.org/2014/testing.slide
- https://www.elastic.co/blog/code-coverage-for-your-golang-system-tests

rminnich, 2017-01-25:
that's quite a disappointment. That said, let's go slow and see if we think of something. Good research however ...

rjoleary, 2017-02-02:
I've done some more thinking and experimentation on the issue. There is a fundamental problem:

- If A executes the child process B and we want the coverage of B to be reflected in A, the [GoCover array](https://blog.golang.org/cover#TOC_4.) must be copied from B to A upon completion of B. Since, `os.Exit` and `log.Fatal` exit the program immediately (all goroutines exit, no defers are called, no finalizes are called), there are many exits from B and no obvious means of making the copy.

I thought of one hacky solution however:

- Create a [test subprocess](https://talks.golang.org/2014/testing.slide#23) for the test. In the subprocess, memory map the [GoCover array](https://blog.golang.org/cover#TOC_4.) to a file, then execute `main`, then `os.Exit(0)`.
- Even if the process dies, the latest version of memory will be mapped to the file even without an msync [[source]](http://stackoverflow.com/a/6219962).
- The parent process reads from the file and copies the contents into its own GoCover array.

It would require some unsafe Go and be fairly fickle.

I believe a better solution is to dynamically invoke terminating functions such as `os.Exit` and `log.Fatal`. You can see an example here: http://stackoverflow.com/a/40801733 .

- The test will set `osExit=func() {panic("exit called")}` and call `main` (no subprocessing). The test can check if `osExit` is called using a `recover`.
- When the program is executed for real, `osExit=os.Exit`.

This second approach might also benefit bbnofork because the terminating functions can be easily caught and not terminate the shell.

rminnich, 2017-02-02:
The second looks good and the terminating functions idea will work. We'll need to document these conventions or people will forget to use them, but then we'll know as our coverage will go to zero.

I'm in CAM today and will try to ask around about other possible ideas.

prattmic, 2017-12-26:
Making coverage work on subprocesses is unfortunately much harder with the go tool than with Bazel.

First, you need to instrument all of the files you care about. Doing a normal build from within go test -cover won't cut it.

     $ go tool cover -mode=set -var=CoveredFoo -o foo_covered.go foo.go

Then somewhere in the subprocess you need to do something like https://github.com/golang/go/blob/master/src/cmd/go/internal/test/test.go#L2020-L2022 and https://github.com/golang/go/blob/master/src/cmd/go/internal/test/test.go#L2047-L2094 to process all of the generated coverage maps.

Finally, before the subprocess exits, you need to get the results back to the parent somehow. You could spit out coverage profile format files (https://github.com/golang/go/blob/master/src/testing/cover.go#L99) then merge them back together in the parent.

[testing.RegisterCover](https://godoc.org/testing#RegisterCover) can be used to replace the coverage data that the test framework will look at when the test completes.

tl;dr, this is all a huge pain. It would be much easier to refactor the commands to allow testing via import instead of subprocess.

hugelgupf, 2017-12-26:
I think in many ways you are right - many of the commands should be refactored a bit. But more importantly, I think we should have more stable, self-contained tests that don't rely on testing against the "upstream" command's implementation. You and I could have different versions of wget on our machines, and `EXECPATH="wget -O -" go test ./cmds/wget` may work for me, but fail for you.

prattmic, 2017-12-26:
Ah, testing against a reference implementation isn't something I considered, but that could still be done by usually calling `wget.Main(args)` vs exec.Command for the reference version.

Differences between different reference is definitely a valid concern, especially considering how many differences the BSD vs GNU tools have, though I think it is a largely different issue.

prattmic, 2017-12-26:
I actually ran into this a few weeks ago when I was trying to cleanup pwd. It turns out GNU pwd (/bin/pwd) and the Bash builtin pwd have different defaults for -L and -P.

hugelgupf, 2017-12-26:
It's a little more like `Main(args, stdin, stdout, stderr)`, which gets to be a little more annoying. 

In addition to that, bb mode's flag rewriting prevents the use of `flag.CommandLine.Parse(args)` instead of os.Args, so that makes things even more annoying. See #534 for the stop-gap for now.

rjoleary, 2017-12-26:
>>> tl;dr, this is all a huge pain. It would be much easier to refactor the commands to allow testing via import instead of subprocess.

I agree. Nothing I saw for the subprocess approach would be easy to implement.

>>> But more importantly, I think we should have more stable, self-contained tests that don't rely on testing against the "upstream" command's implementation.

The original use case for "$EXECPATH" was to compare the performance of several Go commands with similar GNU BusyBox commands. Without a reference implementation, it is hard to know what performance is acceptable. This led to the optimizations of a few commands, especially `dd`.

If you want to use EXECPATH to test functionality, it would probably be best to stick with a single version and config of BusyBox and not use bash builtins.

rjoleary, 2020-12-04:
It looks like this package has developed a way to do it: https://github.com/TheDiveO/gons/blob/master/reexec/testing/testmain.go#L55

Part of this post: https://stackoverflow.com/a/60583196

MedicineYeh, 2020-12-04:
It looks good to me. I will investigate this in the following week.

rjoleary, 2021-09-07:
I think we're going to cave and just implement this the old boring way where `main` calls a separate `run` function which can be easily tested (ex: https://github.com/u-root/u-root/blob/master/cmds/contrib/spidev/spidev.go#L114).

closed 2021-09-17 by rjoleary