u-root/u-root #294: Multiple DHCPv6 Servers in the same network

xchenan, 2017-07-18:
Now the DHCPv6 client just handles the first DHCPv6 packet it gets after it sends out a request. even if there are multiple DHCPv6 servers in the same network. It is okay if both servers reply with a valid IPv6 address.

However, if one server does not have available address for the client whereas the other has, and the former one replies faster than latter, the client will miss the DHCPv6 address that is assigned to it.

A possible solution would be: keep client in the loop until it receives a DHCPv6 reply that has a valid IPv6 address, or time runs out.

rminnich, 2017-07-18:
I think that your last sentence is correct -- that's the way to do it. But does the spec have any suggestions? For example, the ARP spec is very clear about how to handle multiple responders.

hugelgupf, 2024-02-20:
I believe the DHCP client has been doing what the last line suggests for a while. If there is no address to configure, this will print a log statement and we continue waiting for more packets: https://github.com/u-root/u-root/blob/ec531346790ffce37001de26a3412a206869931c/cmds/core/dhclient/dhclient.go#L97

closed 2024-02-20 by hugelgupf