u-root/u-root #312: srvfiles does not work

ananion-cation, 2017-08-04:


hugelgupf, 2017-10-30:
What did you see instead?

rjoleary, 2018-06-27:
I couldn't reproduce @ananion-cation's issue. It could be a networking issue, firewall, forgetting to expose the port in Docker, ...

However, I came across another issue related to caching. Steps to reproduce:

1.  `srvfiles -d . -h 127.0.0.1`
2. Open "http://localhost:8080" in Google Chrome. A 200 (OK) response is received, you can see an index listing files, everything works fine.
3. CTRL-C the server from step 1.
4. Start running a new server on the same port, but a different directory: `srvfiles -d .. -h 127.0.0.1`
5. Refresh the page in Google Chrome. A 304 (NOT MODIFIED) response is received. Erroneously, the index from step 1 is displayed rather than step 4.

It seems as though Chrome is caching the page and Go's file server is telling Chrome to use its cache. It would be better if `srvfiles` never returns a 304 response by default.

go version devel +f040e439cb Tue Jun 26 23:56:13 2018 +0000 linux/amd64

rjoleary, 2018-07-15:
I couldn't reproduce anion-cation's issue. I'll put my new, unrelated findings into #808

closed 2018-07-15 by rjoleary