u-root/u-root #326: flags are not combined in rm, ps, and ls

ananion-cation, 2017-08-08:
You have to split up flags ex: ls -l -R, not ls -lR

rminnich, 2017-10-08:
This will require we use a different package than flag. 

I'm fine with that, but we need to find that packager.

ggiamarchi, 2017-10-09:
Maybe https://github.com/jawher/mow.cli

Probably the best lib I know for this work

tydavis, 2017-12-11:
May be an appropriate replacement: github.com/spf13/pflag

> Package pflag is a drop-in replacement for Go's flag package, implementing POSIX/GNU-style --flags.
>pflag is compatible with the GNU extensions to the POSIX recommendations for command-line options. See http://www.gnu.org/software/libc/manual/html_node/Argument-Syntax.html
>Usage:
>pflag is a drop-in replacement of Go's native flag package. If you import pflag under the name "flag" then all code should continue to function with no changes.

bensallen, 2017-12-11:
We looked at using pflag while developing the new gzip cmd. It's pretty reasonable.

My biggest complaint is it appears it always requires a long flag to be specified when defining the flag, and then provides the option of a short flag. There's plenty of instances where we will just want a short flag.

tydavis, 2017-12-11:
@bensallen I see what you mean. Looks like there's an issue reported for it here: https://github.com/spf13/pflag/issues/139

tydavis, 2017-12-13:
This could be accomplished by adding [a `fallthrough` option](https://golang.org/ref/spec#Fallthrough_statements) for compound short-flags. Also would require some modifications to the [short argument parsing system](https://github.com/spf13/pflag/blob/master/flag.go#L943), including an answer for what happens with repeat arguments (e.g. `-vvv`) and compound forms (e.g. `rm -rf`).

Maybe it's worth forking the flags package (yet again?) and building our own short-options parser.

tydavis, 2017-12-13:
@bensallen Per a comment from `theckman` in the gophers slack #general channel, the [spf13/pflag](https://github.com/spf13/pflag) fork does _not_ require long-form arguments -- based on [the API docs](https://godoc.org/github.com/spf13/pflag), only the functions that end in `P` require the long-form (POSIX) argument.

*OR*

We can [check out gnuflag](https://godoc.org/github.com/juju/gnuflag) which implements what I think we want.

rminnich, 2017-12-13:
same questions: will this work with packages that have their own flags using the flag package? Pretend a package has
flag.BoolVar(...)
will we see that flag if we use gnuflag? 
What happens with busybox mode?

bensallen, 2017-12-13:
I suggest whoever wants to take this on, go through and search/replace `import "flag"` to `import flag "github.com/spf13/pflag"` or `import flag "github.com/juju/gnuflag"` across all cmds and pkgs. Ensure the code generator for BB mode is updated as well: https://github.com/u-root/u-root/blob/master/pkg/uroot/bb.go#L46. Then we simply see what happens...

bensallen, 2017-12-13:
As my comment implies, I think if we're going to move to a new flag package, we should do so holistically. Not a cmd at a time.

hugelgupf, 2017-12-13:
I feel like we have more pressing issues right now, and trying to make it work with bb might be more complicated than we think. And then, you'll be going down the rabbit hole of improving bb...

I guess I'm trying to say this might be a bit of a time sink before bb is more general.

pwFoo, 2018-03-07:
That should be a flag combo problem too?
#508 

```
WARN[2018-03-04T22:53:16.815596551Z] Running modprobe bridge br_netfilter failed with message: flag provided but not defined: -va
Usage of modprobe [-n] modulename [parameters...]:
  -n    Try run
, error: exit status 2 
```

pwFoo, 2018-04-03:
Any updates / plans with the flag combo problem?

hugelgupf, 2018-04-13:
cc @subhav 

Now that I've submitted #555 and #708, bb-mode builds are not tied to the flag package anymore like they used to be. It should be pretty straightforward to use pflag now? Maybe @subhav wants to try it out on a couple commands and see if it still works in both source mode and bb mode compiled initramfs.

hugelgupf, 2018-04-13:
I would, in fact, try a bb mode build with three commands:

```
./u-root -build=bb ./cmds/test1 ./cmds/test2 ./cmds/test3

(or directly use tools/makebb to just make the binary)
```

`test1` would be a simple command using the `flag` package.
`test2` would be a simple command using the `pflag` package.
`test3` would be a simple command using the `gnuflag` package.

`./bb test1 -a`, `./bb test2 -b`, `./bb test3 -c` should all work.

rjoleary, 2018-07-11:
Ron moved the `rm` command to pflag in #791 

We still need to move the rest of u-root's commands to pflag.

mem, 2018-10-05:
> We still need to move the rest of u-root's commands to pflag.

I noticed cp doesn't support flag combos yet. I'd be happy to make a list of the commands not yet using pflag and work on those. For now I'll create a PR with the change for cp which is the one I need and I already wrote. Hopefully that's ok.

closed 2019-09-23 by hugelgupf