u-root/u-root #370: slack channel non-open

tpepper, 2017-09-21:
Joining u-root.slack.com indicates one must have a Princeton or MIT email address.  Can this be opened?

rminnich, 2017-09-21:
we can open it but I also sent you an invite. I'm not very good at slack.

hugelgupf, 2017-09-21:
we should open it up, imho

rminnich, 2017-09-22:
can anyone see how to do that? I only see invite only or selected domains.

ryukinix, 2017-10-05:
In general they use a external tool Ron: https://github.com/outsideris/slack-invite-automation. Which can be easily and free hosted on heroku

hugelgupf, 2017-10-30:
Our slack channel is apparently also reaching the 10,000 message or word limit or whatever the free tier provides.

Any alternatives?

bensallen, 2017-10-31:
Alternatives: 

- https://matrix.org, common client is https://riot.im/app/.
- https://gitter.im
- Good 'ol IRC

ryukinix, 2017-10-31:
Matrix? I like it @bensallen

hugelgupf, 2017-10-31:
I made a Matrix channel to try it out -- #uroot:matrix.org -- try joining and see how you feel about it?

I've used Matrix/Riot for some other things, but haven't been super impressed by the UI. But it's free, it works, and it's kind of like IRC, so I'm not against it.

hugelgupf, 2017-11-02:
For those of you who want to try it out in a browser: https://riot.im/app/#/room/#uroot:matrix.org

rck, 2017-11-03:
In general I prefer async communication like mailing lists/PRs. Other than that I would vote for IRC. That is something I have idling around in some tmux session anyways. I guess matrix/riot should also be fine, I don't really like slack.

hugelgupf, 2017-11-09:
Trial Slack sign-up page: http://slack.u-root.com:3000/

(I need to be a real sysadmin and make that run behind :80, but I've been too lazy.)

bensallen, 2017-11-09:
Worked for me.

hugelgupf, 2017-11-20:
set it up as http://slack.u-root.com now. https coming when i have a chance to get a cert.

tlaurion, 2025-04-01:
https://u-root.org/#community -> https://slack.osfw.dev/

Does not work.

mpictor, 2025-09-18:
Still broken :(

tlaurion, 2025-09-22:
> - https://u-root.org/#community pointing to slack -> https://slack.osfw.dev/
> - https://www.linuxboot.org/  contact section pointing to slack (join here link) -> https://slack.osfw.dev/
> 
>  https://slack.osfw.dev/ does not work.
> 
> Posting this link in both linuxboot and u-root rooms for people in charge to follow up.
> - https://osfw.slack.com/archives/CCW2L82EM/p1743521209559999 (u-root channel)
> - https://osfw.slack.com/archives/C8H5Z29PC/p1743521183578019 (linuxboot channel)

@rminnich ping

hugelgupf, 2025-09-22:
hey @walterchris you were going to set up a website for me to point slack.osfw.dev at. Did you get a chance to?