u-root/u-root #380: Review and clean up our README and other files

rminnich, 2017-10-08:
They need to be looked at by someone with a fresh perspective and any errors fixed.

hugelgupf, 2017-10-08:
I... have always been complaining about the readme; you should remind me to do this when we're all back in town.

(Obv, other new eyes on this would be great, too!)

rck, 2017-10-30:
Yes, I can help with that. I started to contribute 2 days ago, so I still remember the basic questions  I had (e.g., ramfs.go vs. bb) and stuff I had to figure out the "hard" way because it was not documented (e.g., "how to add files to the initramfs"). And I also would have some questions on "how to go from where I'm now".

So if you are up to answer some basic questions newcomers like me have, we can then wrap this up for the README.

If OK, I will write some more structured questions tomorrow. Just committing to this now, before I decide otherwise tomorrow/before I find it out on my own and stuff looks to obvious :-)

hugelgupf, 2017-10-30:
Roland: that would be awesome. I'm also currently working on re-writing scripts/ramfs.go and bb.go into one tool that can create any of the initramfs cpio archives we want in the different configurations. That should make getting started with u-root, NERF, and NiChrome much easier. I'll add you on some code reviews. No worries if not all of it makes sense yet, but if you can tell me if it makes things easier or harder to use that would be awesome!

rminnich, 2017-10-31:
That's a terrific idea!

Get a few PRs in soon and you can get a hacktober shirt :-)

rck, 2017-10-31:
- So what triggered me was the [slides](https://schd.ws/hosted_files/osseu17/84/Replace%20UEFI%20with%20Linux.pdf), especially the backup slides at the end. So two things:
  - Maybe we should put a link to them, or try to get the catchy part in the README
  - If I would have just found a pointer to that repo, I would have asked myself "why should I care?". So again, the slides are good, because they put the project in a bigger picture.

- That is already worked on, but I was confused about the ramfs.go vs. bb situation. Two tools that do the same thing? Why?

- Adding stuff to the initramfs could be a typical usecase (I needed to get a .ko file into it to test the insmod/rmmod/lsmod stuff). So maybe mentioning `bb -add` early would be good.

- The qemu-example just states `path/to/kernel`. We could mention that there are no "special" requirements, and for a start you can use whatever you have in `/boot`.

- At first sight I was a bit overwhelmed with all the options: build an initramfs which contains just the source, or "busybox" mode,... We should mention that people that only want to implement a currently missing program just can switch to `./cmds` and hack right away on their workstation.  

- Maybe add a hint for the more HW focused guys that want to see stuff blink. Is there some "blessed"/cheap dev-platform?

- So cool, I now have my qemu and booted the initramfs, what can I do from now? How would I use `init`? Can I boot to some rootfs? How? (qemu-examples). If not, why not? For example Alpine Linux provides a "mini root fs". What would be required from a checkout of the code to landing on an Alpine shell?

Feedback is welcome. For most points ACK/NACK is enough, I can then update the README accordingly. The HW point and especially the last point require feedback.

rminnich, 2017-10-31:
Got the questions, I have to go to lunch, will answer afterward. I *really* appreciate your patience and perseverance.

rminnich, 2017-10-31:
o That is already worked on, but I was confused about the ramfs.go vs. bb situation. Two tools that o do the same thing? Why?

They both create an initramfs. But the bb tool creates a busybox-mode initramfs, where there is one binary, and lots of symlinks to it, and no source at all. That one binary implements all the commands that bb was asked to compile in when bb was run. The default is all the commands. But things can be changed or added, as determined by a glob:
bb src/github.com/elves/elvish s*/g*/u*/u*/c*
gets all the commands and the elvish shell.

The ramfs tool creates an initramfs containing the go toolchain and an init, and all the source needed for all of u-root. There are no other binaries:  commands are compiled at boot time or when the command is first run.

ramfs was a bash script that grew up to a go program. I added bb in 2015 because we needed a way to make the initramfs smaller and faster, for machines with small flash that are slow. bb is a break with the u-root goal of "see all the source" but we had no choice with space-limited or slow systems.

So, they do different things, for different reasons, but both generate an initramfs.

o Adding stuff to the initramfs could be a typical usecase (I needed to get a .ko file into it to test the o insmod/rmmod/lsmod stuff). So maybe mentioning bb -add early would be good.

Great point! Please do.

o The qemu-example just states path/to/kernel. We could mention that there are no "special" 
o requirements, and for a start you can use whatever you have in /boot.

we need to add one or more actual commands that users use.

o At first sight I was a bit overwhelmed with all the options: build an initramfs which contains just the o source, or "busybox" mode,... We should mention that people that only want to implement a 
o currently missing program just can switch to ./cmds and hack right away on their workstation.

and in fact they don't need to switch to cmds. They can just add the command's go source path and it gets compiled in to bb, which is very convenient.
bb s*/g*/u*/u*/cmds/* s*/github.com/my/repo/
gets them both and they get compiled in to one binary if it's bb; or all needed package source will also be added if it is ramfs.go.

o Maybe add a hint for the more HW focused guys that want to see stuff blink. Is there some 
o "blessed"/cheap dev-platform?

My current favorite: https://www.pcengines.ch/apu2.htm

saturday, I built a new linux kernel from ToT and a new u-root into it as initramfs, and it "just worked", up to and including network! Shall I add a HOWTO? apu2 is very nice and it runs coreboot.

o So cool, I now have my qemu and booted the initramfs, what can I do from now? How would I use o init? Can I boot to some rootfs? How? (qemu-examples). If not, why not? For example Alpine Linux o provides a "mini root fs". What would be required from a checkout of the code to landing on an 
o Alpine shell?

This points to our need for more use cases, but look at github.com/u-root/NiChrome for one example. Here's a movie of NiChrome booting, and mind you, this is mostly u-root
https://photos.app.goo.gl/qHbcUaxPGzpYQbFA3

Also, note, the cpio switch to initramfs was designed to let you merge u-root into an existing initramfs for another distro. We've done this with tinycore and it works. Is this useful with Alpine? No idea. 

I will take a further look.

rck, 2017-10-31:
Thanks for that, I will wrap up what we already have and send a PR tomorrow.

tklauser, 2017-11-11:
> My current favorite: https://www.pcengines.ch/apu2.htm
>
> saturday, I built a new linux kernel from ToT and a new u-root into it as initramfs, and it "just worked", up to and including network! Shall I add a HOWTO? apu2 is very nice and it runs coreboot.

Very nice! A HOWTO on how to get u-root running on the apu2 would be very much appreciated.

closed 2017-12-20 by hugelgupf