u-root/u-root #381: Set up an easy-to-use qemu test harness

rminnich, 2017-10-08:
What we'd like is an easy to use qemu setup. You'll need to find a good place to get a kernel (remember, this test should work on OSX, so using /boot/whatever is not fair!). The steps would be something like
1. go run script/initramfs.go for the architecture you want to test on
2. get a kernel for that architecture that has network, and graphics
3. run qemu-xx-yy with the right arguments

This script should be written in Go, not bash, zsh, or whatever.

rjoleary, 2017-10-21:
You can store kernels and qemu builds in git large file storage (git-lfs) which is also written in Go (https://github.com/git-lfs/git-lfs). It would be wise to place this in a separate repo. However, GitHub might have limits on the bandwidth.

rjoleary, 2018-07-10:
I'm working on this at the moment. Here is the general outline:

1. Run the tests with `go run integration/run_tests.go`. This will start the qemu instances.
2. Each test is its own init script written in Go. For example, one init script might test running the kexec command and another init script runs the shutdown command.
3. Use https://github.com/google/goexpect to verify the output of qemu's serial.

rjoleary, 2018-07-27:
This is down and integrated into circleci with #827

closed 2018-07-27 by rjoleary