u-root/u-root #391: godoc and exports

tpepper, 2017-10-17:
Clean up unnecessary exports and document intended exports so https://godoc.org/github.com/u-root/u-root reads cleaner and CI can gate on linter complaints about exports/comments.

hugelgupf, 2017-10-17:
Part of this is also that I would like to move toward a more packaged
structure. We have commands that use os/exec to call dhclient even though
we have a DHCP package that could be used directly. With this move, what's
exported and not would also be cleaned up (though the two can happen
separately).

rminnich, 2017-10-17:
Chris, overall, I agree with you but dhclient is in my view the wrong choice for an example. dhclient is a persistent process in many cases, and in many of them there is a convention that SIGHUP causes it to reload config files. If we built a dhclient into init, then init would have to figure out what to do with each SIGHUP: what part of init is a SIGHUP for? In the case of dhclient I think a process is the right move. In the case of other things, like cpio, we ought to see if using the package makes sense.

hugelgupf, 2017-10-17:
You're right; bad example.

rjoleary, 2017-10-21:
@tpepper  If you plan on working on this, here is some history/advice/ideas:

* The current format is loosely based on the output of the UNIX `man` command. Originally, the plan was to write our own `man` command in Go - a command which would fetch the "man page" from the command's source code. Since the u-root runtime has access to all the source code and by using Go's ast package, it should be straightforward. We just hadn't had the time.
* https://godoc.org/github.com/u-root/u-root/cmds produces a slightly nicer list of commands than the one you've linked.
* There is a script here which checks the licenses via regex, something similar could be used to check the man pages: https://github.com/u-root/u-root/blob/master/scripts/checklicenses/checklicenses.go
* As for linting, the biggest problem isn't the formatting. Instead it is the list of flags becoming out-of-sync with the flags registered in the source code. This is a bit tricky to check automatically, but you can probably look at the ast to find out which flags are missing. However, some commands (like dd and ip) have non-standard parsers for their args.
* For gating, you might want to look at the pre-commit mechanism: https://github.com/u-root/u-root/blob/master/scripts/pre-commit . Essentially, you add this script to .git/hooks/ and it gates before each commit. It is not automatically setup when you clone a new repository, so this is often overlooked.

hugelgupf, 2019-09-23:
I think we've been moving in the right direction with this, though a lot of `pkg`s still have kind of bad godocs. I'm going to start paying more attention in reviews.

closed 2019-09-23 by hugelgupf