u-root/u-root #392: u-root.tk now points to a domain squatter page

chingcodes, 2017-10-18:
The project url is a dead url ( that shows ads). Did the DNS get renewed?

ryukinix, 2017-10-20:
Sorry for that, I'm the owner of the domain. But I just lost the domain because I forgot to renew and the freenom guys are smart, if you forgot to renew you need to pay the next year. We need a new domain since now is not more free or pay on freenom for the last year. @rminnich is your turn.

[image]

ryukinix, 2017-10-20:
But if we will pay, would be better another domain, like `uroot.org` or `u-root.org`.

hugelgupf, 2017-10-20:
I just went ahead and registered u-root.com.

ryukinix, 2017-10-20:
Yes... too late. But do you not think that `u-root.org` is good enough?

hugelgupf, 2017-10-21:
I'd have preferred .org, but it was taken. Unless you took it?

ryukinix, 2017-10-21:
No, I don't took it. Sorry, I misunderstood the dns seller site that I was looking for alternatives. The `u-root.org` was indeed taken. `.net` is available, although.

rjoleary, 2017-10-21:
I forget the exact steps, but there are a few things needed to rebuild the website with a new URL:

1. The site URL needs changing in the yaml: https://github.com/u-root/u-root/blob/master/website/mkdocs.yml
2. Rebuild the site with the mkdocs tool: http://www.mkdocs.org/
3. This should recreate the /docs folder. Make sure the /docs/CNAME file contains the new URL.
4. Merge this change into master.
5. IIRC github.io will pickup the change and redeploy the site automatically. However, some additional steps might be needed to reconfigure github.io and the DNS servers.

ryukinix, 2017-10-21:
Yes, this was exactly what I did in the first version @rjoleary.

ryukinix, 2017-10-21:
Since someone buy a new domain, we can generate the static website.

rminnich, 2017-10-23:
I bought u-root.tk back. There are way too many talks out there with that name. We have it for 5 years. 
You need to tell me how to set it up.

rminnich, 2017-10-23:
So the plan is we just let u-root.tk die in a few years, we don't reference it any more, but we don't want to let it break. We can't just instantly turn it off, there are too many presentations out there that use it. 
So how do I configure it?

rminnich, 2017-10-23:
why am I not able to reach this page from github.io? What's the URL? I'm ready to do the CNAME dance but ...

rminnich, 2017-10-23:
OK folks it's back. Thanks chingcodes for pointing this out, and thanks to all of you for the discussion. This gives us time to transition with a bit less excitement.

closed 2017-10-23 by rminnich