u-root/u-root #401: KEXEC_FILE only works on X86_64

GanShun, 2017-10-30:
CONFIG_KEXEC_FILE in the linux kernel requires X86_64. 

Right now unfortunately it seems like we're stuck on x86 unless we either use the regular kexec or we port KEXEC_FILE over to the other architectures like arm or powerpc.

hugelgupf, 2017-10-30:
We may have to bite the bullet and write a kexec that uses kexec_load(2). Porting kexec_file_load(2) to arm and/or powerpc looks equally unappetizing, to be honest.

rminnich, 2017-10-31:
trust me, porting kexec_file_load is not the end of the world, it's been done. It's far more desirable than using the horrible kexec_load interface. I've done that.

hugelgupf, 2017-10-31:
Well, kexec_file_load is just a Kernel-Level wrapper around kexec_load. I'm afraid it's gonna be almost the same work either way. But I'll take a look at the kernel interface.

rminnich, 2017-10-31:
yes, but I can tell you from experience that writing to the old model is pretty painful, particularly in Go, as you need to also port the purgatory to build in user mode, since you have to supply it. It's not fun.

The messiness of the old model has limited its use in some companies I know of.

vejmarie, 2017-11-28:
Can somebody try legacy_boot script, I can't kexec currently. I got a load failure. On my distro kexec from kexec-tools is using kexec_load not kexec_file_load, I got a weird error key was rejected by service from my go script

rminnich, 2017-12-05:
same error. But I don't get this error from the kexec command. Something's different.

rminnich, 2017-12-05:
Consider doing what the kexec command does and see if that works.

rminnich, 2017-12-05:
make sure you don't have KEXEC_VERIFY_SIG in .config for kernel

marcoguerri, 2018-05-18:
Hi there, I don't mean to hijack this discussion, so please let me know if you want me to open another thread. I have a question concerning `kexec_load`, in particular for @rminnich . Assuming somebody out there is willing to go through the pain of implementing it in Go for u-root, initially targeting only `x86_64`, would that be valuable for u-root? I am not thinking only about multiarch support, but rather support for multiboot images. My understanding is that `kexec_file_load` does not support multiboot, and I was wondering if, implemented the `kexec_load` way, it would be something that u-root could benefit from.

GanShun, 2018-05-18:
Oh definitely! We've been wanting multiboot support for awhile. That would be really great.

zaolin, 2018-05-23:
@marcoguerri @GanShun https://lwn.net/Articles/747812/

rjoleary, 2019-01-20:
I have a rough draft for kexec-ing a zImage on arm with kexec_load. It fails while decompressing the kernel, which I'll keep debugging. I'll open a PR once it's passing the qemu kexec test.

It's likely we'll need a Go package which can parse and write device tree.

rjoleary, 2021-05-27:
Upstream Linux already has a kexec_file_load for arm64 now (https://github.com/torvalds/linux/blob/master/arch/arm64/kernel/machine_kexec_file.c). It's missing the ability to change the dtb, but that's a separate problem.

closed 2021-05-27 by rjoleary