u-root/u-root #419: Exiting u-root on QEMU through console-mode throws a kernel panic

ryukinix, 2017-11-04:
### Setup

I build the intiramfs via:

`/bb 'src/github.com/u-root/u-root/cmds/[a-z]*' src/github.com/elves/elvish/` on the master commit 

As described on README.md through the currently master commit https://github.com/u-root/u-root/commit/7aae4c44cac481da84c6777f62e2216a15658075 (tried on #404 but is the same behavior)

The command-line for boot qemu is:

```
qemu-system-x86_64 -kernel /boot/vmlinuz-linux-lts -initrd /tmp/initramfs.linux_amd64.cpio -append "console=ttyS0" -nographic
```
Which attach the content on the current terminal instead of opening a GUI graphic (I have problems with my keyboard with QEMU on graphic mode)

## Output
```
❯ qemu-system-x86_64 -kernel /boot/vmlinuz-linux-lts -initrd /tmp/initramfs.linux_amd64.cpio -append "console=ttyS0" -nographic
2017/11/04 00:09:09 Created dir :"/proc": mode 555
2017/11/04 00:09:09 Created dir :"/sys": mode 555
2017/11/04 00:09:09 Created dir :"/buildbin": mode 777
2017/11/04 00:09:09 Created dir :"/ubin": mode 777
2017/11/04 00:09:09 Created dir :"/tmp": mode 777
2017/11/04 00:09:09 Created dir :"/env": mode 777
2017/11/04 00:09:09 Created dir :"/etc": mode 777
2017/11/04 00:09:09 Created dir :"/tcz": mode 777
2017/11/04 00:09:09 Created dir :"/dev": mode 777
2017/11/04 00:09:09 Created dir :"/lib": mode 777
2017/11/04 00:09:09 Created dir :"/usr/lib": mode 777
2017/11/04 00:09:09 Created dir :"/go/pkg/linux_amd64": mode 777
2017/11/04 00:09:09 Created dev :"/dev/tty": mode: 020666: magic: 1280
2017/11/04 00:09:09 Created dev :"/dev/urandom": mode: 020444: magic: 265
2017/11/04 00:09:09 Created dev :"/dev/port": mode: 020640: magic: 260
2017/11/04 00:09:09 Created mount -t "proc" -o  "proc" "/proc" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "sysfs" -o  "sys" "/sys" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "tmpfs" -o  "cgroup" "/sys/fs/cgroup" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "tmpfs" -o  "none" "/tmp" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "devtmpfs" -o  "none" "/dev" flags 0xc0ed0000
2017/11/04 00:09:09 Created dir :"/dev/pts": mode 777
2017/11/04 00:09:09 Created mount -t "devpts" -o newinstance,ptmxmode=666,gid=5,mode=620 "none" "/dev/pts" flags 0xc0ed0000
2017/11/04 00:09:09 Created symlink "/dev/ptmx" -> "/dev/pts/ptmx"
2017/11/04 00:09:09 Created file "/etc/resolv.conf"
2017/11/04 00:09:09 Created dir :"/sys/fs/cgroup/memory": mode 555
2017/11/04 00:09:09 Created dir :"/sys/fs/cgroup/freezer": mode 555
2017/11/04 00:09:09 Created dir :"/sys/fs/cgroup/devices": mode 555
2017/11/04 00:09:09 Created dir :"/sys/fs/cgroup/cpu,cpuacct": mode 555
2017/11/04 00:09:09 Created symlink "/sys/fs/cgroup/cpu" -> "/sys/fs/cgroup/cpu,cpuacct"
2017/11/04 00:09:09 Created symlink "/sys/fs/cgroup/cpuacct" -> "/sys/fs/cgroup/cpu,cpuacct"
2017/11/04 00:09:09 Created mount -t "cgroup" -o memory "cgroup" "/sys/fs/cgroup/memory" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "cgroup" -o freezer "cgroup" "/sys/fs/cgroup/freezer" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "cgroup" -o devices "cgroup" "/sys/fs/cgroup/devices" flags 0xc0ed0000
2017/11/04 00:09:09 Created mount -t "cgroup" -o cpu,cpuacct "cgroup" "/sys/fs/cgroup/cpu,cpuacct" flags 0xc0ed0000
2017/11/04 00:09:09 Run &{/inito [/inito] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42020e000 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
2017/11/04 00:09:09 fork/exec /inito: no such file or directory
2017/11/04 00:09:09 Run &{/ubin/uinit [/ubin/uinit] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42020e0a0 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
2017/11/04 00:09:09 fork/exec /ubin/uinit: no such file or directory
2017/11/04 00:09:09 Run &{/ubin/rush [/ubin/rush] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42020e140 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
% 2017/11/04 00:09:11 rush: Can't open a console; no job control in this session
% exit
[    8.845657] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[    8.845657]
[    8.848875] CPU: 0 PID: 53 Comm: init Not tainted 4.9.59-1-lts #1
[    8.849229] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.10.3-20171021_125229-anatol 04/01/2014
[    8.850262]  ffffc9000017bc30 ffffffff812fa08d ffff880005fee800 ffffffff8191dee8
[    8.850262]  ffffc9000017bcb8 ffffffff8117905d ffff880000000010 ffffc9000017bcc8
[    8.850262]  ffffc9000017bc60 2e8932cf14034e6f ffff880005fee800 0000000000000000
[    8.850262] Call Trace:
[    8.850262]  [<ffffffff812fa08d>] dump_stack+0x63/0x86
[    8.850262]  [<ffffffff8117905d>] panic+0xe4/0x22d
[    8.850262]  [<ffffffff81081a48>] do_exit+0xab8/0xac0
[    8.850262]  [<ffffffff81081acb>] do_group_exit+0x3b/0xb0
[    8.850262]  [<ffffffff8108c691>] get_signal+0x291/0x600
[    8.850262]  [<ffffffff811fc41e>] ? mem_cgroup_commit_charge+0x7e/0x4d0
[    8.850262]  [<ffffffff8102d0a7>] do_signal+0x37/0x6a0
[    8.850262]  [<ffffffff8102c6e2>] ? __switch_to+0x202/0x600
[    8.850262]  [<ffffffff810035fc>] exit_to_usermode_loop+0x8c/0xb0
[    8.850262]  [<ffffffff81003aa9>] syscall_return_slowpath+0x59/0x60
[    8.850262]  [<ffffffff815fd844>] entry_SYSCALL_64_fastpath+0xa7/0xa9
[    8.850262] Kernel Offset: disabled
[    8.850262] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
                        [    8.850262]
```

Why that happens? On ancient `-test=true` of ramfs.go I don't have that problem. @rminnich

rminnich, 2017-11-04:
That's quite normal. When you exit rush, init exits, and init is pid 1. This is standard behavior of linux but we need to make init better (maybe).

ryukinix, 2017-11-04:
@rminnich thanks for pointing that, but you can explain why I need exit twice?

ryukinix, 2017-11-04:
There is another way to running u-root (easily) instead of `QEMU` since the flag `-test` was removed from `ramfs.go`? (just for know)
After the kernel panic my terminal makes unresponsive, C-c doesn't works (problem of QEMU probably).

hugelgupf, 2017-11-04:
There'll be a new test mode with the new tool.

rminnich, 2017-11-04:
I yanked test mode out because it had gotten really hard to correctly test init with control ttys and all that stuff. I think qemu is still the best way to go for test but I'll see what Chris comes up with.

ryukinix, 2017-11-04:
Nice, I'll wait for that. I think we should keep that open until we have this new tool, because someone can pass through for the same thing as I passed.

closed 2017-12-20 by hugelgupf

ryukinix, 2017-12-20:
@hugelgupf there is some improve on that or this was just closed?

hugelgupf, 2017-12-20:
@ryukinix there isn't a new test mode yet, but I think for that there should be a different issue. The kernel panics when init exits - that's just what the kernel does.

ryukinix, 2017-12-20:
@hugelgupf this is enough. Thanks.