u-root/u-root #420: Using bb is really painful

prattmic, 2017-11-05:
The README is not clear at all on how to use it. It says:

```
The following command for example includes the elvish shell:

$ bb src/github.com/elves/elvish s*/g*/u*/u*/c*
```

My attempts:

1. From `$GOPATH` (note my `GOPATH` is `/home/prattmic/prog/opensource/u-root/`):

```
$ go build github.com/u-root/u-root/bb
$ ./bb src/github.com/elves/elvish s*/g*/u*/u*/c*                                                                                                                                                 prattmic@thinkpratt
2017/11/04 17:25:03 Using "/home/prattmic/prog/opensource/go" as GOROOT
cmd_elvish.go:3:8: cannot find package "github.com/u-root/u-root/bb/bbsh/cmds/elvish" in any of:
	/home/prattmic/prog/opensource/go/src/github.com/u-root/u-root/bb/bbsh/cmds/elvish (from $GOROOT)
	/home/prattmic/prog/opensource/u-root/src/github.com/u-root/u-root/bb/bbsh/cmds/elvish (from $GOPATH)
2017/11/04 17:25:04 exit status 1
```

The problem seems to be that the generated `$GOPATH/bbsh/cmd_elvish.go` imports `github.com/u-root/u-root/bb/bbsh/cmds/elvish`, but that package is at `$GOPATH/bbsh/cmds/elvish` not `$GOPATH/src/github.com/u-root/u-root/bb/bbsh/cmds/elvish`.

2. From `$GOPATH/src/github.com/u-root/u-root/bb`:

```
$ go build github.com/u-root/u-root/bb
$ ./bb src/github.com/elves/elvish s*/g*/u*/u*/c*
zsh: no matches found: s*/g*/u*/u*/c*
```

Doh. Of course those paths are wrong now.

```
$ ./bb $GOPATH/src/github.com/elves/elvish ../cmds/*
2017/11/04 17:29:52 Using "/home/prattmic/prog/opensource/go" as GOROOT
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/u-root/home/prattmic/prog/opensource/u-root/src/github.com/elves/elvish matched no paths
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/cmds/ansi matched no paths
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/cmds/archive matched no paths
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/cmds/builtin matched no paths
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/cmds/cat matched no paths
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 =================================================
2017/11/04 17:29:52 Warning: /home/prattmic/prog/opensource/cmds/chmod matched no paths
2017/11/04 17:29:52 =================================================
```

Something is very screwed up here. Are those package paths (with an extra `src/` on the front?), not filesystem paths?

3. Back to package paths, this time letting `bb` do globbing instead of the shell:

```
$ ./bb src/github.com/elves/elvish "s*/g*/u*/u*/c*"
2017/11/04 17:43:41 Using "/home/prattmic/prog/opensource/go" as GOROOT
cmds/elvish/main.go:21:2: cannot find package "github.com/boltdb/bolt" in any of:
	/home/prattmic/prog/opensource/u-root/src/github.com/u-root/u-root/vendor/github.com/boltdb/bolt (vendor tree)
	/home/prattmic/prog/opensource/go/src/github.com/boltdb/bolt (from $GOROOT)
	/home/prattmic/prog/opensource/u-root/src/github.com/boltdb/bolt (from $GOPATH)
2017/11/04 17:43:41 exit status 1
```

`elvish` seems to have this package in `$GOPATH/src/github.com/elves/elvish/vendor/github.com/boltdb/bolt`, but I guess `bb` doesn't understand that? I guess I'll just install it...

```
$ go get github.com/boltdb/bolt
$ ./bb src/github.com/elves/elvish "s*/g*/u*/u*/c*"
2017/11/04 17:46:02 Using "/home/prattmic/prog/opensource/go" as GOROOT
# github.com/u-root/u-root/bb/bbsh
./initmap.go:5:9: undefined: cmdsInit
2017/11/04 17:46:07 exit status 2
```

Uhh, well I have an empty `bbsh/cmds/cmds`. I assume that's wrong. Maybe that glob needs another directory?

4. Fix glob:

```
$ ./bb src/github.com/elves/elvish "s*/g*/u*/u*/c*/*"
2017/11/04 17:51:12 Using "/home/prattmic/prog/opensource/go" as GOROOT
2017/11/04 17:51:12 Can't Parsedir /home/prattmic/prog/opensource/u-root/src/github.com/u-root/u-root/cmds/.gitignore, readdirent: not a directory
Output file is in /tmp/initramfs.linux_amd64.cpio
```

Huzzah! Wow, that was hard. I only figured out (3) and (4) as I wrote this issue.

hugelgupf, 2017-11-05:
Hey Pratt, I'm working on a tool with better UI to replace bb. So it'll be addressed soon :)

rminnich, 2017-11-05:
I would argue that part of the problem is that zsh is badly broken in that every shell since 1971 has passed in a glob if it had no matches. zsh is way to clever for its own good. That said, bb is a pain in the neck, so any improvements are welcome.

prattmic, 2017-11-05:
Sure, but was only one problem. The others were:

1. What directory does bb need to be run from?
2. What arguments should it be passed? (The example glob doesn't work)
3. What dependencies need to be installed? github.com/elves/elvish was obvious, github.com/boltdb/bolt wasn't.

hugelgupf, 2017-11-05:
The new tool will support two ways of specifying packages. As paths relative to your cwd (any cwd, not just bb) and as go import paths (e.g. github.com/u-root/u-root/cmds/ls or github.com/u-root/*).

rminnich, 2017-11-05:
yes, the boltdb thing is weird, I think elvish should be vendoring it but they are not.

rminnich, 2017-11-05:
I'm too tired, you're right, they're vendoring it. I forget why this fails but I know they keep having issues building when
CGO_ENABLED=0
could this be part of it? Oh well I'm beat. Later :-)

rminnich, 2017-11-05:
around july 20 the boltdb went into elvish main.go and things won't work after that. It's not clear why -- the build command finds the vendor path just fine in most cases.

Elvish now scares me. It dops a unix socket at a random place and then you can talk to it. This seems a basic Bad Idea to me. Unless someone can convince me this is not an issue, I think it's time to take elvish out of our README and continue a search for a shell in Go. 

Maybe adding command completion to rush is enough. Or maybe we port pdksh to go. I'm not sure. But elvish is getting insanely complex.

dancrossnyc, 2017-11-05:
It seems to me that a go-native version of ksh93 would be a useful thing. I
wonder if Dave Korn could be roped into the effort?

hugelgupf, 2017-11-20:
Fixed via #418.

closed 2017-11-20 by hugelgupf