u-root/u-root #423: More robust bb source rewriting

prattmic, 2017-11-05:
Currently, for each command to be included in the final binary, bb does the following (please correct me if I missed important steps):

1. Rewrites main to Main.
2. Rewrites init to Init.
3. Creates a command flag.FlagSet `cmdflag`.
4. Rewrites `flag.Foo()` calls to `cmdflag.Foo()`.

This has a number of limitations, ordered from minor to major:

1. It only supports a single init function.

This is trivially fixable.

2. It doesn't rewrite global var assignment.

These may have side effects. This is fairly easily fixed by moving the assignments into Init.

3. It screws up local variables named `flag`.

The AST doesn't know if `flag` is a package or variable identifier.

4. It doesn't rewrite assignments to flag.Usage, breaking custom usage output.

I have a commit to fix this, but it doesn't quite work correctly because a bunch of packages set os.Arg[0] to change the usage output, but flag.FlagSet.Usage does not print os.Arg[0] like flag.Usage does.

5. It doesn't fix up any other global side effects.

This is the fundamental problem.

Imports are not rewritten, so if a command package imports another package which has side effects, it will affect every other command.

That means any flags registered in other packages will end up on `flag.CommandLine` instead of `cmdflags`.

Or that package may have init functions with other arbitrary side effects.

I think this can be solved robustly by doing away with flag-specific rewriting, and instead rewriting all source, including imported packages, to encapsulate all side effects.

For example, a package like this:

```go
package main

import (
    "fmt"

    "import1"
    "import2"
)

var foo = flag.Bool("foo", false, "bar")

func init() {
    fmt.Printf("initing")
}

func main() {
    flag.Parse()
    // Do stuff.
}
```

Becomes this:

```go
package cmd

import (
    "fmt"

    "import1"
    "import2"
)

var foo *bool

func initVars() {
    foo = flag.Bool("foo", false, "bar")
}

func init1() {
    fmt.Printf("initing")
}

var initDone bool

func Init() {
    if initDone {
        return
    }

    import1.Init()
    import2.Init()

    initVars()
    init1()

    initDone = true
}

func Main() {
    flag.Parse()
    // Do stuff.
}
```

Notice that use of the flag package no longer needs to be rewritten, because no flags will be registered (in any package) until Init is called.

The generated init is basically a recreation of the normal Go init functions, recursively calling init on all dependencies.

This approach has a few challenges:

1. We probably don't want to rebuild the standard library, so we can't rewrite it. However I think that is mostly fine. The only package I can think of with side effects on import is "testing" (adds flags).

2. Moving global variable assignment into a function requires type information. The AST alone will no longer suffice. This is needed to fix (3) above anyways.

3. The real main package in this binary needs to be written very carefully to avoid its own side effects, or using these rewritten packages without calling Init.

This is definitely trickier than the existing mechanism, but beyond not rewriting the standard library, I can't think of any cases where it would break. Please let me know what I forgot.

prattmic, 2017-11-05:
I forgot to mention that #418 provides a much nicer interface for building this than the existing bb.go.

hugelgupf, 2017-11-05:
Would you mind holding off on this until the new build script is out? I've already done half of this.

prattmic, 2017-11-05:
Yeah, I haven't done anything and wouldn't want to start without your refactor anyways.

rminnich, 2017-11-06:
So, I'd like to add some perspective to this discussion.

bb is a limited use tool for two limited cases

o limited firmware space. u-root in dynamic compilation mode is about 6M. That is a tough size for older systems with smaller flash. For newer systems, it's not an issue.

o slow hardware. The Allwinner A20 is a dog. It can take a minute or more to boot with u-root if we're compiling. Note that this time is dramatically slower now than with go1.2, and while it's not as bad as 1.6 it's still not great. Go as a language is trending away from its 'super fast compilation' beginnings and u-root relied on super fast compilation. But even back in those days, compilation on ARM was nothing to be excited about.

So the role of bb is to provide a precompiled set of tools for limited ROM space. If the CPUs are slow, but ROM space is huge, there are other options.

If you have unlimited ROM space,or are a root file system, what might we do? We could do use the u-root that is based on source, but precompile all the tools. Then you get the source and the boot speed. 

If you are slightly more limited in ROM space, and your CPUs are really fast, what might you do? What we have done on e.g. systems with 16M ROM, compile on boot.

If you are highly constrained on ROM space, what can you do? That's where bb comes in.

bb is not an end in itself. If you can show, for example, that we could use dynamic linking to solve the problem (dynamic linking for Go did not exist when I started u-root), and you could somehow make all that work in 2M, then we can drop bb. bb is not important. Fitting in 2M is important. If you have lots more ROM, bb makes no sense, unless you are angling for super fast boot time on slow CPUs, and again, maybe precompiling some of the u-root programs is a better move.

So we have knobs we turn. Precompile into one binary, don't precompile, precompile into many binaries. 

Again, bb is not intended to be the common use case. It's a special case, for special cases.

Further, given that it's a special case, for small memory, we should not stress if there are things we can not compile in. It's a shame that we can't (e.g.) compile in elvish easily, but if making a single binary makes an initramfs bigger than 6M, that makes no sense anyway. IOW, if the single binary ever makes an initramfs larger than the dynamic compilation case, we should not use the single binary, we should use dynamic compilation.

So, on to the comments:

>It only supports a single init function.
>This is trivially fixable.

Yes, and we've had fixes, we just have not needed it yet. This has never been a big problem. 

>It doesn't rewrite global var assignment.
>These may have side effects. This is fairly easily fixed by moving the assignments into Init.

Seems fine, as long as we're not fixing a problem just to fix it. We did do this once but hit a problem with it I can't remember.

>It screws up local variables named flag.
>The AST doesn't know if flag is a package or variable identifier.

I'm concerned about the extra complexity this might add to cover a use case we can more easily fix by establishing a rule, mainly, 'you don't get to use flag as a variable name if you are being compiled into busybox'. Sometimes adding huge complexity is not worth it when a simple 'thou shalt not' will do.

> It doesn't rewrite assignments to flag.Usage, breaking custom usage output.
>I have a commit to fix this, but it doesn't quite work correctly because a bunch of packages set 
> os.Arg[0] to change the usage output, but flag.FlagSet.Usage does not print os.Arg[0] like 
>flag.Usage does.

So we stop changing os.Arg[0], right? that was pretty recent and if it's a mistake it's easy to undo. We own this code. If we have to do a global change to make something work better we do it. None of this is sacred.

Now we get to the big one:
>It doesn't fix up any other global side effects.

This is the fundamental problem.

>I think this can be solved robustly by doing away with flag-specific rewriting, and instead rewriting >all source, including imported packages, to encapsulate all side effects.

That's pretty neat and if it can be done it would be great. Maybe this is where the real effort should go. The other bits, to me, are not nearly as important. This is the case where nasty things hide.

rjoleary, 2017-12-06:
We had the same discussion early this year. I even made a working implementation, almost exactly as Michael proposes now:

- https://github.com/rjoleary/u-root/blob/bb-fix/bb/bb.go#L295

The dream was to have a completely general `bb` command. Any list of Go packages can be passed in, written in any coding style their authors desired, and they would all be built into a single binary busybox.

The idea was killed by its complexity, edges cases and time:

1. My implementation took the type information into account, but at the cost of a lot of complexity and headache. Unlike with AST, you could no longer translate a single file because type information might refer to any imported package.The type information is also cached on disk by Go's build system which also made things interesting. Additionally, AST transforms are unpleasant to write and read in Go (please let me know if you know of a good library for Go AST transform!).
2. It is also difficult to anticipate all the edge cases. Sometimes, the variable has a type and no value. Sometimes, there is no type and a value. Sometimes, there is a type and a value. It is also common to have variable lists with multiple variables per `var` keyword or defining two variables on the same line. You'd have to dig into the language spec to get all the edge cases and new Go releases might add additional syntax.
3. There were more useful things to do at that time.

I came to a conclusion much less glamourous than the "completely generic bb command". Ron came to this conclusion way before I did.

If you want bb to be reasonably maintainable and understandable, it needs to have a small number of simple transformation rules. Right now, there are only a handful. If you have too many transformation rules, but less than 100% complete, then it is difficult to maintain and understand.

But this introduces a tradeoff in u-root: the Go subset. It has worked very well so far. There is a script which statically checks that each package has at most one init and some other properties so you don't mess up too badly.

I still however think there is a place for the "completely generic bb command". If it is truly general, other projects outside of u-root could also benefit from it immediately.

rminnich, 2017-12-07:
The current bb is admittedly half-assed. It's definitely a problem. 

But if you're going to add a huge amount of complexity to cover all cases, well, you better cover all cases. It can be much more frustrating to work with a tool that covers "all the cases save yours" than a tool that covers "some fraction of useful cases"

So I'm open to a better tool, but it is very important to get it right. We've struggled with that.

The old lesson from Unix applies: sometimes it's better not to try to do it all, but make it clear what you don't do. There were all kinds of ways the old Unix tools had limitations, and the philosophy was "live in the limitations.". DMR once called out the BSD folks for "being too general." Their error? Long file names.

There is where the current bb tool fails IMHO: it's not really clear about what it can't do. 

But we all assume and accept certain limitations without thinking of it. Here's an example: would you support creating a file system that had filenames up to 32G long? If not, why not? If so, why? This actually came up on the Plan 9 list once.

We've moved NiChrome off of bb because we decided the extra 20 seconds of boot time was worth the benefit, namely, having a whole go toolchain there. Hence bb is back to its original purpose: supporting machines with small flash or poor Go compilation performance or both.

Should we just let people know what the limits of bb are, and have a build tag to mark things that can be safely built with bb?

What's the answer? I agree with Ryan, Michael, and Chris, that it would be nice to make a general tool. My only caution is that if you're going to claim full generality, you've really got to do it, and so far the proposals I've seen are along the lines of "add lots of complexity to get us a bit closer but not all the way there." Is that enough? I don't know.

PR #555 (2018-02-11): pkg/uroot: rewrite bb to be more general
Fixes #423.

Still a work in progress, but seems to work with all of our existing ./cmds/* tools. (It compiles, I haven't tried it yet. But I'm taking that as a good sign...)

The downside is that bb mode now takes longer to create an initramfs than source mode. But it's still manageable.

cc @prattmic @rminnich @rjoleary
closed 2018-02-11 by hugelgupf