u-root/u-root #424: Size flag handling

rck, 2017-11-06:
Currently we don't really have a nice way to handle size flags. I wrote some flags compatible lib for that. It needs some polishing and obviously tests, but it works pretty well, including showing the default on "-h" and so on. It was written with "truncate" in mind, so it also handles/exposes an explicitly given sign.

https://github.com/rck/size/blob/master/size.go

It would be used like this:
```
var mapping size.Mapping = size.Mapping{
...
        "KiB": 1024,
....
}

var flagS = size.Flag("s", size.NewSize(42, "KiB", mapping), "Parse size including unit")
```

I had a version that provides a "default mapping", where the user just specified a "filter" like `[]string{size.KiB,...}`, but then I saw the crazy dd size flags like 'c' and 'w' and decided that it is the callers responsibility to provide a full mapping. Could be changed...

So some questions:
- Do we want that/something similar?
- If so, where would we put it?

If there is some interest and you tell me where it should land, I would prepare a PR with it + tests and "truncate" as the first user. When that is in, I would then look where we can use it as well (dd is an obvious candidate).

rminnich, 2017-11-06:
seems quite cool, I'm curious about other's comments.

hugelgupf, 2017-11-07:
Hey! I like the idea. There are a couple changes I would make. I'll send you a PR for some of them.

I'd also call the package `units` or `unit` instead of `size`. We can use your package and vendor it in, if you like.

hugelgupf, 2017-11-07:
Sent https://github.com/rck/size/pull/1

rck, 2017-11-23:
This took a bit longer than expected... Thanks to Chris for the good feedback. I fixed some minor things on top of that, the result is now available at:
https://github.com/rck/unit

ACKs/Feedback is welcome, I would then prepare a PR that vendors that in and uses it for the usual candidates (dd, truncate, ...). If it needs changes, please send PRs to the new repo.

PR #488 (2017-12-10): Vendor in unit package and make truncate the first user
This was discussed in: https://github.com/u-root/u-root/issues/424

Again, thanks to Chris for his feedback.

If this goes in, I then start git-grepping the tree to find other users for this package.

Closes #424
closed 2017-12-10 by rminnich