u-root/u-root #437: shutdown command: implement "standard" flags and behavior

hugelgupf, 2017-11-10:
Missing:

```
shutdown -h
shutdown -r
shutdown -H
shutdown -P
shutdown -c
shutdown TIME
```

rminnich, 2017-11-11:
here's the rule:

if command has the same name, it has to roughly act the same
if it has a switch that matches an existing tool, that switch must act the same
it can have a switch not found in the standard tool
it doesn't have to implement all the switches

The options to shutdown are just a mess, so let's figure this out carefully, esp. since testing it is tricky.

hugelgupf, 2017-11-11:
I figured they wouldn't be too hard, but I'm fine with just having -h and -r work the same way.

shwoop, 2017-11-23:
I thought this would be a good issue to pick up as I familiarise myself with the codebase but after adding some tests to validate the input I've consulted what shutdown does on sysv/systemd and it appears to be very dumb: `-hr` will reboot, `-hrhPH` will halt, and so forth.

So, I'm guessing that in the interests of mirroring existing systems I should go for the simple dump option rather than any fancy argument validating system?

rminnich, 2017-11-23:
There's no requirement, ever, that you mirror existing systems. If existing systems are stupid then we don't want to mirror them. Shutdown is a very old command (think DEC 32-bit minicomputers,1978 or so, where I first saw it) and it's not aged well. 
Let's not recreate 40-year-old mistakes.

Thanks for looking at this. 
You might rename shutdown to something else and then make the arguments sane. 
We're trying to avoid being dragged into legacy silliness. So if you'd like, create a new command, with sane options, and a different name than shutdown, and let's go with that. 

IOW, if shutdown is too insane to implement in a clean way, or implementing the standard results in ugly Go code, then let's not do it.

hugelgupf, 2017-11-23:
I feel like the only two shutdown flags I've ever actually wanted and needed are `shutdown -h` for halt and `shutdown -r` for reboot. Maybe we should just require you use one or the other and be done with it.

rminnich, 2017-11-23:
yes. We really want to be careful about implementing all of GNU flags or POSIX flags. They are frequently badly thought ought or tied to legacy issues that no longer exist.

shwoop, 2017-11-23:
Thanks for the clarification.  Clearly I read too much into the reviews of #467 (which was my first introduction to the project) - specifically the suggestion that we stick close to coreutils which I interpreted as current linux ehavior.

I'm apprehensive about augmenting `shutdown` with a new command, picking a name alone breaks me out in a sweat.

What about extending shutdown so have a usage of:
```
shutdown [-dryrun] [-k] -h|-r|[halt|reboot|suspend]
```
The validation isn't too complicated and it's backwards compatible.

rminnich, 2017-11-23:
remember, if we do something we don't like, we change it. I assume your syntax is actualy
-[h|r|reboot|suspend]
?
or did you want something like
shutdown halt
which I also like