u-root/u-root #440: u-root should Glob

rminnich, 2017-11-11:
The problem with u-root counting on the shell globbing is that programs, like the usb command in NiChrome, are now required to run an intermediate shell to get the globbing done, or do the globbing themselves, ... it's all kind of annoying. I don't personally view this change as an improvement. I can deal with it, I just don't much like it. 

How does a Go program in, e.g., NiChrome, invoke the u-root command? Please don't tell me to run an intermediate shell. We've avoided dependence on a particular shell in u-root for many years and I don't want to reintroduce that sort of thing. The intermediate shell starts to bring in all kinds of problems of its own, such as its search for . files, and lots of other environment, that may not exist in a pared down configuration.

rminnich, 2017-11-11:
well I can manage, so I'm closing this.

closed 2017-11-11 by rminnich

hugelgupf, 2017-11-11:
Another Go program should invoke u-root by importing the u-root package and calling uroot.Build. I messed up where I put Build, so let me move that real quick :)

rminnich, 2017-11-11:
OK, I'll wait to see.

PR #441 (2017-11-14): pkg/uroot: define the command-line interface as a Go API.
This enables people to include pkg/uroot in their own packages and be
met with the same API available on the command-line.

Fixes #440.
commit 6ed1cabb (2017-11-14): pkg/uroot: define the command-line interface as a Go API.

This enables people to include pkg/uroot in their own packages and be
met with the same API available on the command-line.

Fixes #440.