u-root/u-root #442: Website

hugelgupf, 2017-11-11:
Hey Ron, how is the website currently hosted? On your Linode?

Maybe we should migrate to github.io pages. No hosting and easy to maintain.

rminnich, 2017-11-11:
it's hosted on github. uroot.tk just points to it.

hugelgupf, 2017-11-13:
How can I point u-root.com at it as well?

bkochendorfer, 2017-11-13:
It's probably best to have `u-root.com` redirect to `uroot.tk`. That way there is one source of knowledge and you can hold down multiple TLDs.

rminnich, 2017-11-13:
I agree.

hugelgupf, 2017-11-20:
`u-root.com` redirects to `u-root.tk` now that I've finally set up nginx on that server.

Ron, did you happen to accidentally claim `u-root.com` as a Google Apps domain? I was gonna add it as an alias to my existing account, but Google protested that it already existed elsewhere. I'm happy to verify yours via DNS.

closed 2017-11-20 by hugelgupf