u-root/u-root #463: dynamic mode: confusion about who builds

hugelgupf, 2017-11-20:
Both /init and /installcommand have the capability of building commands. /init builds /installcommand, and somewhere along the way, all the symlinks are installed.

This could be easier:
  * the initramfs *could* contain the symlinks (but need not). they only take a couple bytes, though.
  * maybe /init and /installcommand could be the same thing? they should at least both build commands in the same way. this is especially important once we figure out what slows down builds on bare metal vs qemu, as the fix should apply to every build we do.

rminnich, 2017-11-20:
The bare metal build issue seems to be ok now.

The initramfs _could_ contain the symlinks, as busybox mode does (or did, does the new one?) and it is only a few bytes per. They might be better done as a hard link, actually, since that's only one inode. But in firmware, the phrase "they only take a couple bytes" is not something you want to hear. I think we need to measure alternatives. Sometimes you get down to counting bytes. And we need the Symlink creating code in /init for some of the junk in /dev, so it's not going away. That said, I still think a pre-created hard link may be the most compact way to do this. Our CPIO package may make this easy.

/init is probably going to grow in capability. For NiChrome, I'm thinking of adding an (optional at build time) http port to it so we can have a nice landing page when Chrome starts up. Something that has a nice welcome message, lets us graphically set up wifi, upspin, and so on. Just a thought. But init and installcommand are really very different things, which is why I hesitate a bit to merge them.

rjoleary, 2018-08-10:
Was this fixed by #856 ?

closed 2019-09-23 by hugelgupf