u-root/u-root #483: dd_test.go uses more than 4GB of memory

bensallen, 2017-12-04:
```
circleci@256477e6eebf:/go/src/github.com/u-root/u-root/cmds/dd$ /usr/bin/time go test -v -installsuffix uroot -ldflags '-s'
=== RUN   TestDd
signal: killed
FAIL	github.com/u-root/u-root/cmds/dd	10.172s
Command exited with non-zero status 1
12.02user 12.31system 0:14.31elapsed 170%CPU (0avgtext+0avgdata 4172400maxresident)k
136776inputs+26432outputs (38major+2432073minor)pagefaults 0swaps
```

The test case that appears to cause the high memory usage:

```
{
		// 1 GiB zeroed file in 1024 1KiB blocks.
		flags:  []string{"bs=1048576", "count=1024", "if=/dev/zero"},
		stdin:  "",
		stdout: strings.Repeat("\x00", 1024*1024*1024),
},
```

hugelgupf, 2017-12-04:
Hmm. I'll profile dd. It shouldn't use that much at once. Maybe a buffer pool can help us.

bensallen, 2017-12-04:
The test itself is capturing stdout (1GB of zeros) from dd `out, err := cmd.Output()`, as well as generating 1GB of zeros via: `stdout: strings.Repeat("\x00", 1024*1024*1024),`. Then comparing the two:

```
if string(out) != tt.stdout {
	t.Errorf("Want:\n%#v\nGot:\n%#v", tt.stdout, string(out))
}
```

 Without actually profiling anything I'm guessing, but I think the test itself is the source of high memory usage not dd.

rjoleary, 2017-12-06:
@bensallen is right. I measured the RSS high water mark of both processes:

* `dd.test`: 4907700 kbytes
* `dd`: Varies depending on the test, but the maximum is 672360 kbytes.

Both values are large, but the test is much larger.

The test can be solved by reading stdin in chunks.

PR #493 (2017-12-06): cmds/dd: Rewrite dd_test to lower memory use
- Stream stdout of from dd via cmd.Stdoutpipe()
- Add compare function and count fields to the table tests struct to allow for two compare functions to be used.
- For tests that check stdout against a sequential set of identical bytes, create func byteCount. Fixes having to allocate large slices to compare stdout against.
- For tests that need string like comparison, create func stdoutEqual, which compares stdout a byte at a time to the input []byte
- Use byte slices instead of strings where possible
- Don't output error messages including all of st [... cut]
closed 2017-12-06 by bensallen