u-root/u-root #487: Support PXE boot

hugelgupf, 2017-12-04:
We should have a "generic" PXE boot client script that kind of does the following:

1) bring up NIC
2) get DHCP lease (includes IP & TFTP server location -- we should also support some kind of non-TFTP location)
3) download kernel & initramfs from TFTP server (optionally, we should support non-TFTP locations)
4) optionally, gpgv verify the kernel & initramfs
5) kexec the kernel appropriately

bensallen, 2017-12-04:
- HTTP and TLS should also be supported for transfer of the kernel and initramfs. Other than legacy environments TFTP really shouldn't be used for PXE booting anymore.
- Copying a subset ipxe's configuration / script format might be a good start

hugelgupf, 2017-12-04:
I am def going to support http and TLS, but I thought PXE itself only supports TFTP, so for compatibility with current systems we should support it?

insomniacslk, 2017-12-04:
UEFI 2.5+ supports HTTP booting instead of TFTP but previous versions still need TFTP. There are still large non-2.5-compliant deployments so I guess dropping TFTP at the moment is not an option

bensallen, 2017-12-04:
Yes I would suggest supporting both protocols. 

Parse the DHCP boot filename option, if not a URI, default it to TFTP relative to the next-server option. Otherwise DHCP should pass tftp://<server>/path or http[s]://<server>/path. U-root if its not already should provide a user-class or vendor class option identifying itself as "u-root", so dhcpd can pass different filenames based on who is asking (i.e. UEFI, u-root, iPXE).

hugelgupf, 2017-12-04:
Sounds good.

rminnich, 2017-12-05:
however, for us at google it's an option. First we need the basic stuff, and tftp is for certain something we have no intention of wanting or supporting. So, yeah, it's a issue that we might not do tftp first, but should not be a blocker (for googlers anyway).
tfpt is destined to rot in a dark place.

bensallen, 2017-12-05:
I'm perfectly happy with the project not supporting TFTP until someone actually asks for it. Pulling from my notes, Go TFTP libraries:

- https://github.com/pin/tftp
- https://github.com/vcabbage/go-tftp
- https://github.com/vmware/gotftp

Thinking more about a configuration file format, instead of copying iPXE, using a basic JSON config file seems like a good starting point. dhcpd should pass a URL to the remote configuration file as the boot filename option when a client identifies itself as "u-root".

```
{ 
  "kernel": { "URI": "<uri>", "cmdline": "console=ttyS0 ..." },
  "initrd": { "URI": ["<uri1>", "<uri2>", "<uriN>"] }
}
```

A nice feature in iPXE, is being able to specify a URL for the configuration file like `http://10.0.0.1/ipxe.cfg/{hwaddr}`, which is interpreted to the client's hwaddr. Example [dhcpd-template.conf](https://github.com/warewulf/warewulf3/blob/development/provision/etc/dhcpd-template.conf#L15). This is nicer than having a predefined search pattern of paths fetched like pxelinux.

hugelgupf, 2017-12-05:
Thanks for all the pointers. It'd have taken me a couple days to pull all this together :)

hugelgupf, 2017-12-05:
The initial support is going to be for something that works with Google's setup, with the addition of anything that's easy to do (JSON config and TFTP via a library should be easy).

closed 2019-09-23 by hugelgupf

tanyan02, 2021-04-26:
@hugelgupf Hi, can u-root support pxeboot with qemu now?