u-root/u-root #496: New Priority: packages

hugelgupf, 2017-12-07:
It kind of hit me as we were talking about init scripts in Go at the presentation today. If our intended solution for init is a Go program, it becomes all the more important for us to write as many of the more complicated tools in u-root as reusable and modular packages. An (firmware-embedded, not for nichrome) init script should be nothing other than a bunch of package imports and calls to well-defined methods that we already have. The tools should be good enough that it isn't more than like 100 lines of Go. E.g.

```
func main() {
  if kernel, params, err := netboot.TryPXEBoot(); err != nil && err != netboot.ErrNoLease { 
   // Something went v wrong.
   ... Error handling ...
  } else if err == nil { 
    // DHCP gave us a kernel.
    if err := gpg.Verify(kernel, key); ...
    if err := kexec.FileLoad(kernel, nil, params); ...
    if err := kexec.Reboot(); ...
  }

  // No DHCP lease, no netboot. Boot from disk.
  if err := mountDisk(); ...
  if g, err := grub.FindConfig("/dev/sda"); ...
  if err := g["default"].Boot(); ...
```

We should slowly move towards putting more things in packages. Users can still in their init script decide to call out to a subprocess to perform some task (e.g. if they have tools that they trust more than these), but it should be *easiest* and *secure by default* to use u-root packages.

hugelgupf, 2017-12-07:
Maybe this could also mean that a production deployment of a u-root initramfs *is* just one binary: init. If it pulls all the dependencies one needs to init things for the real kernel, what else do you need? (I bet Ron is going to tell me this is a terrible idea.)

bensallen, 2017-12-07:
Fully agree that the majority of the cmds' code should be moved into packages. This would also be a good time to cleanup exports.

I think it also makes sense to define a new flagset in the package with all the CLI flags for the command, and then make use of it in the cmd's main(). This should let BB mode rewriting of flags be a lot cleaner.

I suspect a few design patterns (maybe even an interface or two) should be defined as best practices, eg.

```
type CliCmd interface {
	FlagSet() flag.FlagSet
        ...
}
```

I've also added a proposal around the implementation of init and services in #60.

hugelgupf, 2017-12-07:
I was once in favor of what you propose there. I'd do it in a heartbeat if the purpose of bb was to only include u-root commands.

But one of the goals of u-root is to be able to include any Go program. This is why bb is the way it is currently. We should possibly revisit this decision, but I do think bb can be done in a general way the way it was proposed in another issue earlier.

rminnich, 2017-12-07:
It's a fine line. It's been done, i.e. "it's all in one binary." It has downsides, which is that it can be a bit confusing when you do a ps or go to change one command and have to recompile everything. My main advice: be careful. 

bb should be able to include any command as Chris points out. We don't want to become a closed-off world.

Finally, I see the role of bb decreasing in importance over time, as flash gets bigger, we get more cores, and so on. It's possible bb is something we leave behind at some point, but who knows?

so, packages yes, but make sure the packages can be used elsewhere, and that we can easily import commands, and that we can use them in the dynamic mode, as we are using wingo and upspin today. Sorry, it's late, I'll stop here, can someone review my pci command?

rminnich, 2017-12-07:
oh right forgot. Another goal of u-root is that people should always be able to change a program in simple ways. If you go too far toward burying everything in a package that gets harder. So it's important to maintain the ability of newcomers to easily, at any time, and especially after booting, mess around with things like, e.g., dd and see what happens. Compiling it all into /init would frustrate that goal.

rminnich, 2017-12-07:
and if you bury too much in packages, it's suddenly harder to change how one program works without breaking others. So think carefully on this. It's not near as simple as it looks.

rminnich, 2017-12-07:
ok, just got off the plane, was a little unclear on what you're saying, you're not saying all u-root commands should be compiled into one /init, you're saying all init programs should be compiled into one /init, right? For the dynamic case? You'll still need to deal with some separate programs, so that won't ever go away. Just wanted to mention that. ttyl.

rminnich, 2017-12-07:
sorry, clicked wrong button.

hugelgupf, 2017-12-07:
> ok, just got off the plane, was a little unclear on what you're saying, you're not saying all u-root commands should be compiled into one /init, you're saying all init programs should be compiled [...]

None of the above was quite what I meant. I mean that once you figure out what you want your uinit to do, and it's fully debugged, and you want to deploy u-root in a production setting, why do you need any u-root commands other than uinit as init? Why does anything other than `init` need to be in flash?

> If you go too far toward burying everything in a package that gets harder. So it's important to maintain the ability of newcomers to easily, at any time, and especially after booting, mess around with things like, e.g., dd and see what happens.

What I'm proposing above with `init` does not change the existence of any u-root commands. Everything still exists as standalone commands. I just want most of the logic to move into packages. So `dd` shouldn't end up being much more than a `main` that makes calls to a `parallelio` package or something like that.

This is not too complicated if the packages are only one or two levels deep. Our package setup is fairly simple right now. Code just moves from `cmds/` into `pkg/`, that's not too hard to deal with for beginners.

hugelgupf, 2017-12-07:
> and if you bury too much in packages, it's suddenly harder to change how one program works without breaking others. So think carefully on this. It's not near as simple as it looks.

This is what we need better tests for. Moving things into packages makes things more modular, and more modular code allows us to write more well-defined unit tests for that code.

rminnich, 2017-12-07:
chris, point taken, makes sense, I should not read stuff at 1am after flying/driving.

hugelgupf, 2019-02-08:
I think we're fully in this one. :)

closed 2019-02-08 by hugelgupf