u-root/u-root #497: Investigate better compression algorithms

hugelgupf, 2017-12-07:
We were told today that there are newer and better compression algorithms than lzma that the kernel supports for initramfs. We should look into that, since size is important for us.

bensallen, 2017-12-07:
Arjan from ClearLinux did some good research into comparing the algorithms: https://clearlinux.org/blogs/linux-os-data-compression-options-comparing-behavior.

For best compression size xz or the new zstd (not added to kernel yet for initrd it appears: http://lists-archives.com/linux-kernel/28937359-add-x86-zstd-kernel-compression.html).

hugelgupf, 2019-09-23:
We've been using xz at Google for a while, but in general u-root doesn't dictate compression anyway. Closing this issue since this is a beyond u-root issue.

closed 2019-09-23 by hugelgupf