u-root/u-root #505: move to circle ci

rminnich, 2017-12-10:
this means:
turn off use of travis
have circle ci able to block commits
cp travis.sh to test.sh so people know what to run locally

we'll leave travis files in the repo a bit longer until we're certain circle ci is working for us.

bensallen, 2017-12-11:
Other TODOs:

- Update the badge in README.md:
```
[![CircleCI](https://circleci.com/gh/u-root/u-root.svg?style=svg)](https://circleci.com/gh/u-root/u-root)
```

- Test CircleCI CLI to run locally: https://circleci.com/docs/2.0/local-jobs/. This requires docker.

rminnich, 2017-12-11:
so this docker thing is a problem. What were they thinking?

bensallen, 2017-12-11:
It appears with the move to CircleCI's 2.0 API, which introduced use of Docker (instead of only native VMs) and workflows, they've not finished some functionality specifically if you use workflows. Workflows are what lets us run up-to 4 containers at once, and how we've split our various testing tasks into parallel jobs.

So far I've found with workflows, CircleCI doesn't support capturing junit style test results into the "Insight" dashboard [[1]](https://circleci.com/docs/2.0/collect-test-data/), and the CLI tool doesn't support workflows [[2]](https://circleci.com/docs/2.0/local-jobs/#circleci-command-line-interface-cli-overview). Both of these would be nice to have, but not blockers in my opinion. 

The circleci CLI does let you run individual jobs from the workflow, eg. `circleci build --job dep`. This is pretty slick, it uses Docker, grabs the same container it would use in the cloud and replicates pretty much exactly what you see as output from the website.

Also of interest, the CircleCI CLI is actually running the circle/picard docker container itself. If you're running with SELinux enabled you need to futz with it to allow a container to control Docker and in my case my GOPATH is shared via /vagrant mounted via vmhgfs:

```
allow svirt_lxc_net_t container_runtime_t:unix_stream_socket connectto;
allow svirt_lxc_net_t vmblock_t:file ioctl;

allow svirt_lxc_net_t vmblock_t:file { open read };
allow svirt_lxc_net_t vmblock_t:lnk_file read;
```

rminnich, 2017-12-11:
well it's clearly way better, if more complex, I plan to do the cutover next monday (can't do it this week)

rminnich, 2017-12-31:
It's done, thanks to Ben.

closed 2017-12-31 by rminnich