u-root/u-root #508: Investigate Container Runtimes

bensallen, 2017-12-11:
u-root needs a container solution

- runc (https://github.com/opencontainers/runc) can be compiled with `CGO_ENABLED=0 make BUILDTAGS=''`, making it likely something we can include into u-root. runc can run exported docker containers: https://github.com/opencontainers/runc#creating-an-oci-bundle.

Older notes from roadmap.md:

Options:

* Docker
* Rkt - https://github.com/rkt/rkt/
* wget + unpack (cpio? tar?) + u-root pflask
* implement a gitclone command and use u-root pflask

hugelgupf, 2017-12-12:
I personally would prefer something that uses `runc` since the OCI spec seems to be the standard that the industry is converging on.

SvenDowideit, 2017-12-19:
I might play with this for my Linux.conf.au talk - and yes, `runc` is a good place to start - I was using it to bootstrap to a containerd container, which can then be used as the backend to a dockerd container :)

rminnich, 2017-12-19:
Great. We could really use the help.

pwFoo, 2017-12-23:
@SvenDowideit 
I would love to see some slides or a howto to use i-root as minimal dockerd host.

pwFoo, 2018-01-25:
Anyone played with runc here?
Quick test I get an error which is new to me.

[image]

hugelgupf, 2018-01-25:
Looks like a libcontainer error. I'll be able to take a look tonight.

pwFoo, 2018-01-27:
Tested again with a new runc build, but same error.
Anyone tested runc oder docker running on u-root?

SvenDowideit, 2018-01-28:
Now that I've done my talk, and come back from linux.conf.au, I've got a little more free time, so I'll poke this - I droped it for the talk in favour of showing things I could make work quickly :)

pwFoo, 2018-02-10:
How to mount additional cgroups with the u-root mount command? I tried for example
```
mkdir /sys/fs/cgroup/pids
mount -t cgroup pids -o pids /sys/fs/cgroup/pids
```

Error:
```
Mount :devices: on :-o: type :cgroup: flags c0ed0000: no such file or directory
```

Isn't it supported with mount command?
How to mount cgroups with shell?

rminnich, 2018-02-10:
This invocation of mount seems wrong.
mount -t cgroup pids -o pids /sys/fs/cgroup/pids
in the docs I have it's
mount -t cgroup -o pids /sys/fs/cgroup/pids

rminnich, 2018-02-10:
oops I mean
mount -t cgroup -o pids none /sys/fs/cgroup/pids

pwFoo, 2018-02-11:
Thanks, my shell script used mount -n which is not supported, so I converted it to the example above.
Modified your eyxample to be the same as cgroup mounts during boot. Successful tested.
```
mkdir -p /sys/fs/cgroup/net_cls,net_prio
mount -t cgroup -o pids cgroup /sys/fs/cgroup/net_cls,net_prio flags 0x0

mkdir -p /sys/fs/cgroup/hugetlb
mount -t cgroup -o hugetlb cgroup /sys/fs/cgroup/hugetlb flags 0x0

mkdir -p /sys/fs/cgroup/pids
mount -t cgroup -o pids cgroup /sys/fs/cgroup/pids flags 0x0

mkdir -p /sys/fs/cgroup/blkio
mount -t cgroup -o blkio cgroup /sys/fs/cgroup/blkio flags 0x0

mkdir -p /sys/fs/cgroup/perf_event
mount -t cgroup -o perf_event cgroup /sys/fs/cgroup/perf_event flags 0x0
```

Tested with downloaded runc and self build binary, but same error. I can't find any information about the error message.
```
container_linux.go:265: starting container process caused "process_linux.go:348: container init caused \"rootfs_linux.go:45: preparing rootfs caused \\\"Error found less than 3 fields post '-' in \\\\\\\"36 35 0:5 / /proc rw,relatime - proc  rw\\\\\\\"\\\"\""
wait: exit status 1
```

bensallen, 2018-02-11:
Looking into this a bit, that error is coming from:

https://github.com/opencontainers/runc/blob/a618ab5a0186905949ee463dbb762c3d23e12a80/libcontainer/mount/mount_linux.go#L69

parseMountTable() above it is calling parseInfoFile() to parse /proc/self/mountinfo.

The entry from your error is missing the mount source field. It should look more like:

```
19 62 0:3 / /proc rw,nosuid,nodev,noexec,relatime shared:5 - proc proc rw
```

Try remounting /proc with `mount -t proc proc /rpoc`

bensallen, 2018-02-11:
Actually its more than just proc, try a build with #617's changes.

pwFoo, 2018-02-17:
Tested with latest u-root. I get a new error.
```
~/> runc run --no-pivot --bundle /example/alpine/ alpine
time="2018-02-17T09:42:15Z" level=error msg="open /dev/null: permission denied
" 
open /dev/null: permission denied
open /dev/null: permission denied

[...]

container_linux.go:348: starting container process caused "process_linux.go:402: container init caused \"read init-p: connection reset by peer\""
Exception: runc exited with 1
```

```
~/> ls -l /dev/null 
crw-rw-rw- 0 0 1, 3 Feb 17 09:34 null
```

pwFoo, 2018-02-17:
Second test with **build=bb** and **created a /etc/hosts file**.

```
~/> runc run --no-pivot --bundle /example/alpine/ alpine
/ # cat /etc/os-release 
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.7.0
PRETTY_NAME="Alpine Linux v3.7"
HOME_URL="http://alpinelinux.org"
BUG_REPORT_URL="http://bugs.alpinelinux.org"
```

pwFoo, 2018-02-18:
Tested docker @ runc and it fails / stops with:
```
% runc run --no-pivot docker
time="2018-02-18T13:37:11.157356555Z" level=warning msg="could not change group /var/run/docker.sock to docker: group docker not found" 
time="2018-02-18T13:37:11.161235111Z" level=info msg="libcontainerd: new containerd process, pid: 9" 
time="2018-02-18T13:37:11.174573479Z" level=warning msg="containerd: low RLIMIT_NOFILE changing to max" current=1024 max=4096 
time="2018-02-18T13:37:12.192546478Z" level=info msg="[graphdriver] using prior storage driver: overlay2" 
time="2018-02-18T13:37:12.195778192Z" level=info msg="Graph migration to content-addressability took 0.00 seconds" 
time="2018-02-18T13:37:12.197105613Z" level=warning msg="Unable to find blkio cgroup in mounts" 
time="2018-02-18T13:37:12.197948447Z" level=warning msg="Unable to find cpuset cgroup in mounts" 
time="2018-02-18T13:37:12.198004098Z" level=warning msg="mountpoint for pids not found" 
time="2018-02-18T13:37:12.199859416Z" level=info msg="Loading containers: start." 
time="2018-02-18T13:37:12.253195169Z" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferred IP address" 
Error starting daemon: Error initializing network controller: Error creating default "bridge" network: Failed to program NAT chain: Failed to inject DOCKER in PREROUTING chain: iptables failed: iptables --wait -t nat -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER: iptables: No chain/target/match by that name.
 (exit status 1)
wait: exit status 1
```

All modules (iptables, nat, ...) have to be loaded manually. Also with deps. Looks like modprobe doesn't solve deps itself?
```
% lsmod
Module                  Size  Used by
ip_set                 40960  0
ebtable_nat            16384  0
ebtable_filter         16384  0
ebtable_broute         16384  0
ebtables               32768  3 ebtable_nat,ebtable_filter,ebtable_broute
ext4                  614400  0
jbd2                  102400  1 ext4
mbcache                16384  1 ext4
iptable_security       16384  0
ipt_MASQUERADE         16384  1
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
nf_conntrack_netlink    40960  0
nfnetlink              16384  2 ip_set,nf_conntrack_netlink
iptable_nat            16384  1
nf_nat_ipv4            16384  1 iptable_nat
nf_conntrack_ipv4      16384  2
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
iptable_raw            16384  0
e1000                 143360  0
iptable_filter         16384  1
xt_conntrack           16384  0
nf_nat                 32768  2 nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_conntrack          135168  7 ipt_MASQUERADE,nf_nat_masquerade_ipv4,nf_conntrack_netlink,nf_nat_ipv4,nf_conntrack_ipv4,xt_conntrack,nf_nat
libcrc32c              16384  2 nf_nat,nf_conntrack
overlay                73728  0
ip_tables              24576  4 iptable_security,iptable_nat,iptable_raw,iptable_filter
```

I think there are some more modules missing to start docker inside of a runc container with u-root.

pwFoo, 2018-03-04:
Looks like a modprobe missing option `flag provided but not defined: -va`.

```
% dockerd
WARN[2018-03-04T22:53:15.717762602Z] could not change group /var/run/docker.sock to docker: error parsing groups for docker: open /etc/group: no such file or directory 
INFO[2018-03-04T22:53:15.721070268Z] libcontainerd: new containerd process, pid: 173 
WARN[0000] containerd: low RLIMIT_NOFILE changing to max  current=1024 max=4096
ERRO[2018-03-04T22:53:16.746706326Z] 'overlay' not found as a supported filesystem on this host. Please ensure kernel is new enough and has overlay support loaded. 
ERRO[2018-03-04T22:53:16.761449136Z] 'overlay' not found as a supported filesystem on this host. Please ensure kernel is new enough and has overlay support loaded. 
ERRO[2018-03-04T22:53:16.767459271Z] devmapper: Udev sync is not supported. This will lead to data loss and unexpected behavior. Install a more recent version of libdevmapper or select a different storage driver. For more information 
INFO[2018-03-04T22:53:16.799478451Z] Graph migration to content-addressability took 0.00 seconds 
WARN[2018-03-04T22:53:16.801509762Z] Your kernel does not support swap memory limit 
WARN[2018-03-04T22:53:16.803188265Z] Unable to find blkio cgroup in mounts        
WARN[2018-03-04T22:53:16.804758742Z] Unable to find cpuset cgroup in mounts       
WARN[2018-03-04T22:53:16.806347943Z] mountpoint for pids not found                
INFO[2018-03-04T22:53:16.808175290Z] Loading containers: start.                   
WARN[2018-03-04T22:53:16.815596551Z] Running modprobe bridge br_netfilter failed with message: flag provided but not defined: -va
Usage of modprobe [-n] modulename [parameters...]:
  -n    Try run
, error: exit status 2 
WARN[2018-03-04T22:53:16.825446104Z] Running modprobe nf_nat failed with message: `flag provided but not defined: -va
Usage of modprobe [-n] modulename [parameters...]:
  -n    Try run`, error: exit status 2 
WARN[2018-03-04T22:53:16.835032439Z] Running modprobe xt_conntrack failed with message: `flag provided but not defined: -va
Usage of modprobe [-n] modulename [parameters...]:
  -n    Try run`, error: exit status 2 
Error starting daemon: Error initializing network controller: error obtaining controller instance: failed to create NAT chain: Iptables not found
wait: exit status 1
```

pwFoo, 2018-04-07:
Running alpine with runc was sucessful two month ago. Running dockerd should be possible if flag combos (#326) issue ist solved. Dockerd try to load modules (iptables) with unsupported modprobe option `-va`.

pwFoo, 2018-04-15:
@hugelgupf 
Will dockerd need some more mounts?
```
WARN[2018-04-15T06:56:20.731294337Z] Unable to find blkio cgroup in mounts        
WARN[2018-04-15T06:56:20.732510289Z] Unable to find cpuset cgroup in mounts       
WARN[2018-04-15T06:56:20.733747114Z] mountpoint for pids not found        
```

Should be added to boot / init?

Tested with quick and dirty modified `modprobe -va` (#712)

pwFoo, 2018-05-20:
I have tested runc containers (mdevd, dhcpcd, rngd) successfully with u-root built initrd (u-root "native" rootfs and an alpine based rootfs).

Runc containers work fine!

Still a problem with iptables nat for docker with native u-root rootfs. So runc works without problems. To run docker or a docker runc container in u-root the iptables nat / modprobe problem should be fixed. I have no idea what the problem could be.

pwFoo, 2018-08-24:
Is there a fix to run containers without the alpine rootfs (modprobe,...) dependency?

pwFoo, 2018-08-26:
nf_nat bug still exists with latest release. So dockerd can't start until iptables nat doesn't work.

pwFoo, 2018-08-29:
Fixed iptables problem. Binary works fine. modprobe too!
Dockerd isn't loading all the needed modules! Some modules need to be preloaded!
```
modprobe -va nf_conntrack_ipv4 iptable_nat iptable_filter ipt_MASQUERADE xt_addrtype veth
```

bensallen, 2018-09-13:
Another runtime we should investigate running on u-root: https://github.com/containers/libpod

pwFoo, 2018-10-28:
Latest u-root with elvish shell docker is easy to start 
```
modprobe veth <NIC_DRIVER>
dhclient -retry 1 -renewals 1 <DEV>

E:DOCKER_RAMDISK=true
dockerd --iptables=false &

docker run --rm -ti --net=host alpine sh # --net=host is needed without iptables for network connection
```

pwFoo, 2019-03-12:
[Podman](https://github.com/containers/libpod) as runtime?

mupuf, 2023-05-09:
FYI, I created [boot2container](https://gitlab.freedesktop.org/mupuf/boot2container) as a standalone initramfs. It is based on `u-root`, `podman`, `minio-client`, and some other projects.

 The container runtime is controlled through the kernel command line, so that computers can be netbooted with whatever container pipeline a user may want.

tlaurion, 2023-05-11:
@mupuf any reason why ttys0 output of qemu is not redirected to host console from launched docker? Would be really nice here! (Trying to test under qemu+heads here, not real hardware :) )

mupuf, 2023-05-12:
> @mupuf any reason why ttys0 output of qemu is not redirected to host console from launched docker? Would be really nice here! (Trying to test under qemu+heads here, not real hardware :) )

That would be due to the fact that you set 2 `consoles=` in the commandline, with the last one not being `console=ttyS0`. Please see https://gitlab.freedesktop.org/mupuf/boot2container/-/issues/48 for more information about it :) Feel free to propose any solution to this problem there!