u-root/u-root #511: TestPsExecution Unrelated Test Failed

bensallen, 2017-12-11:
https://circleci.com/gh/u-root/u-root/223 failed due to unrelated work of the PR:

--- FAIL: TestPsExecution (0.00s)
	ps_test.go:18: Loading Table fails on some point; read /proc/11209/stat: no such process
FAIL
coverage: 27.8% of statements
FAIL	github.com/u-root/u-root/cmds/ps	0.015s

Taking a wild guess, is `ps` iterating over the directory /proc/<pid> and bailing if a process disappears before its completes?

rjoleary, 2017-12-11:
This is a recurring issue and has been fixed in the pass. It likely started again due to a regression with recent changes to `ps`.

Previous bug and fix: https://github.com/u-root/u-root/issues/28 and https://github.com/u-root/u-root/pull/168

Can you reproduce locally by running the following two commands in two separate shells?

* `while true; do /bin/true; done`
* `for i in {1..1000}; do ./ps > /dev/null; done`

This stresses `ps`'s ability to iterate over nodes as they change.

hugelgupf, 2019-09-23:
These kinds of race conditions *look* to have been fixed at some point: https://github.com/u-root/u-root/blob/ed07a650de4cbdab16929540e87a1d6f0c2c613e/cmds/core/ps/ps_linux.go#L250

closed 2019-09-23 by hugelgupf