u-root/u-root #513: source build mode: race between building rush and init

hugelgupf, 2017-12-12:
the commands init tries to run after setting up stuff are these:

```go
cmdList := []string{"/inito", "/buildbin/uinit", "/buildbin/rush"}
```

however, uinit and rush are actually being built in the background by installcommand. if installcommand hasn't gotten to rush and uinit by the time init looks for them, init will just exit (and then panic your kernel).

hugelgupf, 2017-12-12:
This is actually not quite the case, since there's a symlink from /buildbin/rush to installcommand, but combined with issue #514 I think what happens is that the build of rush fails for some reason.

rjoleary, 2017-12-12:
This shouldn't be the issue because these commands are all symlinks to installcommand. In the worst case, the command will be built twice, but there shouldn't be a failure.

hugelgupf, 2017-12-12:
It fails combined with #514 :)

hugelgupf, 2017-12-12:
Weirdly enough, it all works locally in a chroot.

hugelgupf, 2017-12-12:
i'm going to assume that "rush" is actually found, but there is some kind of build error as in #514. Closing for now.

closed 2017-12-12 by hugelgupf