u-root/u-root #514: source mode: installcommand errors with "exit status: 1"

hugelgupf, 2017-12-12:
at least in my setup it comes up twice: 

```
2017/12/12 02:08:23 installcommand error: exit status 1
2017/12/12 02:08:25 installcommand error: exit status 1
2017/12/12 02:082017/12/12 02:08:26 exit status 1
```

but there is no indication of which command failed to build or why at all. it doesn't impact the commands i use, it seems (rush works?), but it'd be good to know what failed. i assume sooner or later i'll find what it is anyway...

rjoleary, 2017-12-12:
It comes from this line:

    https://github.com/u-root/u-root/blob/10da32c018cbad03d5b27c2336fe650207a78e08/cmds/init/bgbuild.go#L47

This means installcommand failed to build two of the commands. More verbose logging is in order.

You can compare ubin to buildbin to find out which commands weren't built.

hugelgupf, 2017-12-12:
```
2017/12/12 02:47:04 installcommand: trying to build {cmdName: ip, PATH /go/bin/linux_amd64:/go/bin:/go/pkg/tool/linux_amd64:/ubin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/bin:/usr/local/sbin:/buildbin:/bb
in, err exit status 1, out go install runtime/internal/atomic: build output "/go/pkg/linux_amd64/runtime/internal/atomic.a" already exists and is not an object file
}

2017/12/12 02:47:09 installcommand ed error: exit status 1
2017/12/12 02:47:09 installcommand: trying to build {cmdName: ed, PATH /go/bin/linux_amd64:/go/bin:/go/pkg/tool/linux_amd64:/ubin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/bin:/usr/local/sbin:/buildbin:/bb
in, err exit status 2, out # sync
go/src/sync/pool.go:9:2: can't find import: "runtime"
}

% 2017/12/12 02:47:16 installcommand u-root error: exit status 1
2017/12/12 02:47:16 installcommand: trying to build {cmdName: u-root, PATH /go/bin/linux_amd64:/go/bin:/go/pkg/tool/linux_amd64:/ubin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/bin:/usr/local/sbin:/buildbin
:/bbin, err exit status 1, out can't load package: package github.com/u-root/u-root: no Go files in /src/github.com/u-root/u-root
}
```

we are probably missing some dependencies (i assume hfiles, gofiles, and sfiles isn't enough?)

the last thing is probably the command glob. see #418. we need to check for package main.

hugelgupf, 2017-12-12:
i'm wondering if the runtime error is a go compiler concurrency race. (file gets compiled by one or both at the same time and accessed by both at the same time?) just leaving that here for now since it's only happened to me once.

rminnich, 2017-12-12:
it is a go compiler issue. The bgbuild is not without issues. We also want to make it default to off, since on small memory systems like ARM it's not a good idea to build a bunch of 2M commands we might never use ...

rminnich, 2017-12-31:
I made bgbuild disabled by default, so this problem seems to be gone.
People can experiment with bgbuild by setting uroot.bgbuild in the kernel cmdline.

closed 2017-12-31 by rminnich