u-root/u-root #525: No network interface / kernel modules

pwFoo, 2017-12-19:
I booted up u-root bb build with qemu, but there is no network interface (only unconfigured "lo") and no network drivers.

Is there some more documentation how to get a working network interface up with dhcp? Maybe additional modules are needed ("/lib/" is empty)?

hugelgupf, 2017-12-19:
Our current model for the embedded systems use case is that you should compile the necessary network drivers into your kernel.

For NiChrome and other workloads, we may go for a modular kernel, but we don't currently have modprobe or anything like udev. So the immediate advice would be to compile it directly into the kernel, still.

pwFoo, 2017-12-19:
Thanks! I used a default desktop kernel without included modules. So I have to compile a custom kernel first.

pwFoo, 2017-12-19:
@hugelgupf 
Do you know a pre-compiled (desktop?) kernel with modules included to download for testing? Otherwise I have to compile a custom one.

rck, 2017-12-19:
FWIW, we should have `insmod` and `lsmod`, I worked on that some time ago.

pwFoo, 2017-12-19:
Hi @rck , thanks for helping here! :)
I tried insmod with a added module (-files, module was included) but it doesn't worked for me. Maybe because the module was .ko.gz instead of .ko?
Is there some more documentation about module loading? Is it possible to include directories (/lib/modules/*)?

hugelgupf, 2017-12-26:
I'm not sure about compression of modules, but modules are very specific to the kernel version you are at. A module built at 4.9.1 may not work with a 4.9.2 kernel. Make sure the module you're including works with that specific kernel.

pwFoo, 2017-12-26:
Kernel and module are the same version from my desktop linux, but modules are compressed .ko.gz. 
Inside u-root gunzip the file failed. Maybe I should gunzip the module first, than include it to u-root and test again to insmod it.
At the moment I'm interested to run containers (#508) on top of u-root.

pwFoo, 2018-01-24:
Used existing kernel and modules to test modprobe / insmod, but with added `/lib/modues` boot failed.
```
/go/src/github.com/u-root/u-root # qemu-system-x86_64 -kernel /kernel -initrd /i
nitrd -append "console=ttyS0" -nographic
e1000: Writing to register at offset: 0x00002410. It is not fully implemented.
e1000: Writing to register at offset: 0x00002418. It is not fully implemented.
e1000: Writing to register at offset: 0x00002420. It is not fully implemented.
e1000: Writing to register at offset: 0x00002428. It is not fully implemented.
e1000: Writing to register at offset: 0x00002430. It is not fully implemented.
e1000: Writing to register at offset: 0x00003410. It is not fully implemented.
e1000: Writing to register at offset: 0x00003418. It is not fully implemented.
e1000: Writing to register at offset: 0x00003420. It is not fully implemented.
```

build command.
```
u-root -files "/lib/modules/:lib/modules/ /usr/bin/runc:usr/bin/runc" -build=bb -o /initrd ./cmds/*
```

rck, 2018-01-24:
What were the commands that were the reason for the printed error messages? A quick `git grep` in u-root and linux did not show these strings. Do your modules and kernel even match? How does `runc` come into the picture?

pwFoo, 2018-01-24:
Runc is just added as binary. Just for fun and test it  later If it boot to a shell. 
Kernel and modules match. Modules copied to lib/modules. Looks like u-root try to load modules during boot? Got the error message after start qemu.
Without modules added it boots to the shell.

hugelgupf, 2018-01-25:
I don't have any ideas right now, but FYI u-root doesn't try to load any kernel modules at the moment.

pwFoo, 2018-01-25:
Mh, looks like a qemu bug / debugging.
I have to take a look later...

insomniacslk, 2018-01-25:
For a kernel with everything built-in you can use Slackware's [huge.s](https://mirrors.slackware.com/slackware/slackware64-current/kernels/huge.s/). Vanilla kernel, 8.2 MB currently, but it's OK for Qemu testing.

insmod etc are working - my old Qemu setup comprised `e1000.ko` and I just used to `insmod` it (until I made a better setup). You can do the same until you find out what's wrong with /lib/modules (and if you don't want to use `huge.s`)

insomniacslk, 2018-01-25:
@pwFoo just discussed a similar issue with a couple of colleagues, that could be the same that has hit you. Does changing this line[1] into `Symlink{NewPath: "/lib", Target: "/usr/lib"},` fix this for you?

[1] https://github.com/u-root/u-root/blob/master/pkg/uroot/util/root.go#L129

pwFoo, 2018-01-25:
Booting kernel+initrd with my dell notebook works fine. And modprobe works too.

@insomniacslk 
Thanks, I have to test it.

pwFoo, 2018-02-18:
With latest u-root build it works fine with qemu and hw.

closed 2018-02-18 by pwFoo