u-root/u-root #531: Investigate strange coverage numbers

hugelgupf, 2017-12-26:
go test -cover does not seem to collect coverage on everything we want. E.g. take

go test -cover ./cmds/wget

which gives 0%, yet there is a test there that covers a bunch of stuff.

Someone should investigate how go test -cover gets its coverage numbers and fix our usage of it or fix go test -cover.

hugelgupf, 2017-12-26:
I'm gonna guess it has something to do with the `testutil` package.

hugelgupf, 2017-12-26:
@prattmic you deal with coverage all the time, if you have 5 mins can you take a guess and tell us what's wrong?

rjoleary, 2017-12-26:
Happy boxing day!

Is this a duplicate of https://github.com/u-root/u-root/issues/116 ?

If it is, our tools are not smart enough to fix it.

prattmic, 2017-12-26:
This is definitely a dup of #116.

closed 2017-12-26 by prattmic