u-root/u-root #533: Turn on golint in u-root

hugelgupf, 2017-12-26:
Whoever takes this on is likely going to have to fix *a lot* of issues in the code base. Documentation alone will likely make up several hundred to thousand lines of changes.

hugelgupf, 2017-12-26:
It should not be mandatory, by the way, but it should be prominent enough to make you fix some things. E.g. sometimes, ALL_CAPS names are fine in Go (e.g. if you are taking it from the Linux kernel), but golint will complain about that.

```
$ golint ./cmds/... | wc -l
232
$ golint ./pkg/... | wc -l
118
```

(this is not in master, but in my pxeboot branch, so numbers may vary)

rjoleary, 2021-05-27:
golint has been enabled in circleci for a while.

closed 2021-05-27 by rjoleary