u-root/u-root #537: 32-bit arm build failure

macat, 2017-12-27:
I was trying to compile u-root for arm 32-bit architecture with `GOARM=7` set, but ran into these issues:

```
GOARCH=arm GOARM=7  u-root -build bb

...
error building go package github.com/u-root/u-root/bbsh: # github.com/u-root/u-root/pkg/gpt
../pkg/gpt/gpt.go:176:83: constant 6075990659671082565 overflows int
# github.com/u-root/u-root/bbsh/cmds/ip
cmds/ip/ip.go:139:23: constant 4294967295 overflows int
cmds/ip/ip.go:144:20: constant 4294967295 overflows int
# github.com/u-root/u-root/bbsh/cmds/shutdown
cmds/shutdown/shutdown.go:31:3: constant 3489725666 overflows int
# github.com/u-root/u-root/bbsh/cmds/switch_root
cmds/switch_root/switch_root.go:34:17: invalid operation: pathFS.Type != fsType (mismatched types int32 and int64)
cmds/switch_root/switch_root.go:134:36: cannot use rootFS.Type (type int32) as type int64 in argument to littleDoctor
, exit status 2
```

Is 32-bit not supported? I saw only arm64 build in circleci build.

rminnich, 2017-12-27:
believe it or not you're the first person in 6 years to try it1

I welcome any and all changes to fix 32-bit, we would really appreciate it!

rminnich, 2017-12-27:
I have a PR up for fixing this. see what you think.

rminnich, 2017-12-27:
darn. I thought that would close this one.

closed 2017-12-27 by rminnich

macat, 2018-01-02:
Thanks @rminnich. This was amazingly fast!

rminnich, 2018-01-02:
we're not from the government but we're here to help anyway.