u-root/u-root #565: README example source build fails to boot

SvenDowideit, 2018-01-15:
on master, I switched from `bb` to `source`, and my qemu then fails to get to rush

`u-root -format=cpio -build=source -o initramfs.cpio`

results in 

```
[    4.824187] Write protecting the kernel read-only data: 18432k
[    4.845932] Freeing unused kernel memory: 2028K
[    4.952109] Freeing unused kernel memory: 1144K
[    5.021160] clocksource: Switched to clocksource tsc
2018/01/15 00:08:27 Welcome to u-root
2018/01/15 00:08:28 Created dir "/buildbin" (mode 0777)
2018/01/15 00:08:28 Created dir "/ubin" (mode 0777)
2018/01/15 00:08:28 Created dir "/tmp" (mode 0777)
2018/01/15 00:08:28 Created dir "/env" (mode 0777)
2018/01/15 00:08:28 Created dir "/tcz" (mode 0777)
2018/01/15 00:08:28 Created dir "/lib" (mode 0777)
2018/01/15 00:08:28 Created dir "/usr/lib" (mode 0777)
2018/01/15 00:08:28 Created dir "/go/pkg/linux_amd64" (mode 0777)
2018/01/15 00:08:28 Created dir "/etc" (mode 0777)
2018/01/15 00:08:28 Created file "/etc/resolv.conf" (mode 0644)
2018/01/15 00:08:28 Created dir "/proc" (mode 0555)
2018/01/15 00:08:28 Created mount -t "proc" -o  "" "/proc" flags 0x0
2018/01/15 00:08:28 Created mount -t "tmpfs" -o  "" "/tmp" flags 0x0
2018/01/15 00:08:28 Created dir "/dev" (mode 0777)
2018/01/15 00:08:28 Created dev "/dev/tty" (mode 020666; magic 1280)
2018/01/15 00:08:28 Created dev "/dev/urandom" (mode 020444; magic 265)
2018/01/15 00:08:28 Created dev "/dev/port" (mode 020640; magic 260)
2018/01/15 00:08:28 Created mount -t "devtmpfs" -o  "" "/dev" flags 0x0
2018/01/15 00:08:28 Created dir "/dev/pts" (mode 0777)
2018/01/15 00:08:28 Created mount -t "devpts" -o newinstance,ptmxmode=666,gid=5,mode=620 "" "/dev/pts" flags 0x0
2018/01/15 00:08:28 Created symlink "/dev/ptmx" -> "/dev/pts/ptmx"
2018/01/15 00:08:28 Created dir "/dev/shm" (mode 0777)
2018/01/15 00:08:28 Created mount -t "tmpfs" -o  "tmpfs" "/dev/shm" flags 0x0
2018/01/15 00:08:28 Created dir "/sys" (mode 0555)
2018/01/15 00:08:28 Created mount -t "sysfs" -o  "sys" "/sys" flags 0x0
2018/01/15 00:08:28 Created mount -t "tmpfs" -o  "cgroup" "/sys/fs/cgroup" flags 0x0
2018/01/15 00:08:28 Created dir "/sys/fs/cgroup/memory" (mode 0555)
2018/01/15 00:08:28 Created dir "/sys/fs/cgroup/freezer" (mode 0555)
2018/01/15 00:08:28 Created dir "/sys/fs/cgroup/devices" (mode 0555)
2018/01/15 00:08:28 Created dir "/sys/fs/cgroup/cpu,cpuacct" (mode 0555)
2018/01/15 00:08:28 Created symlink "/sys/fs/cgroup/cpu" -> "/sys/fs/cgroup/cpu,cpuacct"
2018/01/15 00:08:28 Created symlink "/sys/fs/cgroup/cpuacct" -> "/sys/fs/cgroup/cpu,cpuacct"
2018/01/15 00:08:28 Created mount -t "cgroup" -o memory "cgroup" "/sys/fs/cgroup/memory" flags 0x0
2018/01/15 00:08:28 Created mount -t "cgroup" -o freezer "cgroup" "/sys/fs/cgroup/freezer" flags 0x0
2018/01/15 00:08:28 Created mount -t "cgroup" -o devices "cgroup" "/sys/fs/cgroup/devices" flags 0x0
2018/01/15 00:08:28 Created mount -t "cgroup" -o cpu,cpuacct "cgroup" "/sys/fs/cgroup/cpu,cpuacct" flags 0x0
qemu-system-x86_64: terminating on signal 15 from pid 6163 (killall)
```

pwFoo, 2018-01-26:
My source build boots up slow (qemu without kvm at the moment), but boots to the shell after some time... It feels like minutes.

Each command I use is really slow. Looks like shell stops, but after some time (after command is compiled) it will be executed.

@SvenDowideit 
Did you kill qemu or is it dying? Have you wait some minutes first?

pwFoo, 2018-01-28:
Tested with my laptop it works fine

hugelgupf, 2018-02-10:
If you're running with qemu, you may want -enable-kvm -- compiling from source without hw support is going to be painful. Alternatively, use bb mode to build.

closed 2019-09-23 by hugelgupf