u-root/u-root #567: build fails when adding a go package import with a dash in it

SvenDowideit, 2018-01-15:
```
u-root \
        -format=cpio \
        -build=bb \
        -o initramfs.cpio \
        ./cmds/* \
        github.com/erroneousboat/slack-term \
        github.com/elves/elvish \
        github.com/SvenDowideit/u-root/_examples/uinit
2018/01/16 10:29:42 Disabling CGO for u-root...
2018/01/16 10:29:42 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/OSM/HOME-MEL/dow184/go CGO_ENABLED=0
2018/01/16 10:29:50 error building uroot.BuildOpts{Env:golang.Environ{Context:build.Context{GOARCH:"amd64", GOOS:"linux", GOROOT:"/usr/local/go", GOPATH:"/OSM/HOME-MEL/dow184/go", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}}, Packages:[]string{"github.com/u-root/u-root/cmds/ansi", "github.com/u-root/u-root/cmds/archive", "github.com/u-root/u-root/cmds/boot", "github.com/u-root/u-root/cmds/builtin", "github.com/u-root/u-root/cmds/cat", "github.com/u-root/u-root/cmds/cbmem", "github.com/u-root/u-root/cmds/chmod", "github.com/u-root/u-root/cmds/cmp", "github.com/u-root/u-root/cmds/comm", "github.com/u-root/u-root/cmds/console", "github.com/u-root/u-root/cmds/cp", "github.com/u-root/u-root/cmds/cpio", "github.com/u-root/u-root/cmds/date", "github.com/u-root/u-root/cmds/dd", "github.com/u-root/u-root/cmds/dhclient", "github.com/u-root/u-root/cmds/dirname", "github.com/u-root/u-root/cmds/dmesg", "github.com/u-root/u-root/cmds/echo", "github.com/u-root/u-root/cmds/ectool", "github.com/u-root/u-root/cmds/ed", "github.com/u-root/u-root/cmds/false", "github.com/u-root/u-root/cmds/field", "github.com/u-root/u-root/cmds/find", "github.com/u-root/u-root/cmds/fmap", "github.com/u-root/u-root/cmds/free", "github.com/u-root/u-root/cmds/freq", "github.com/u-root/u-root/cmds/gpgv", "github.com/u-root/u-root/cmds/gpt", "github.com/u-root/u-root/cmds/grep", "github.com/u-root/u-root/cmds/gzip", "github.com/u-root/u-root/cmds/hexdump", "github.com/u-root/u-root/cmds/hostname", "github.com/u-root/u-root/cmds/id", "github.com/u-root/u-root/cmds/init", "github.com/u-root/u-root/cmds/insmod", "github.com/u-root/u-root/cmds/installcommand", "github.com/u-root/u-root/cmds/io", "github.com/u-root/u-root/cmds/ip", "github.com/u-root/u-root/cmds/kexec", "github.com/u-root/u-root/cmds/kill", "github.com/u-root/u-root/cmds/lddfiles", "github.com/u-root/u-root/cmds/ln", "github.com/u-root/u-root/cmds/losetup", "github.com/u-root/u-root/cmds/ls", "github.com/u-root/u-root/cmds/lsmod", "github.com/u-root/u-root/cmds/mkdir", "github.com/u-root/u-root/cmds/mkfifo", "github.com/u-root/u-root/cmds/mknod", "github.com/u-root/u-root/cmds/mm", "github.com/u-root/u-root/cmds/modprobe", "github.com/u-root/u-root/cmds/mount", "github.com/u-root/u-root/cmds/msr", "github.com/u-root/u-root/cmds/mv", "github.com/u-root/u-root/cmds/netcat", "github.com/u-root/u-root/cmds/pci", "github.com/u-root/u-root/cmds/pe", "github.com/u-root/u-root/cmds/pflask", "github.com/u-root/u-root/cmds/ping", "github.com/u-root/u-root/cmds/printenv", "github.com/u-root/u-root/cmds/ps", "github.com/u-root/u-root/cmds/pwd", "github.com/u-root/u-root/cmds/pxeboot", "github.com/u-root/u-root/cmds/readlink", "github.com/u-root/u-root/cmds/rm", "github.com/u-root/u-root/cmds/rmmod", "github.com/u-root/u-root/cmds/run", "github.com/u-root/u-root/cmds/rush", "github.com/u-root/u-root/cmds/seq", "github.com/u-root/u-root/cmds/shutdown", "github.com/u-root/u-root/cmds/sleep", "github.com/u-root/u-root/cmds/sort", "github.com/u-root/u-root/cmds/srvfiles", "github.com/u-root/u-root/cmds/stty", "github.com/u-root/u-root/cmds/switch_root", "github.com/u-root/u-root/cmds/sync", "github.com/u-root/u-root/cmds/tail", "github.com/u-root/u-root/cmds/tcz", "github.com/u-root/u-root/cmds/tee", "github.com/u-root/u-root/cmds/true", "github.com/u-root/u-root/cmds/truncate", "github.com/u-root/u-root/cmds/umount", "github.com/u-root/u-root/cmds/uname", "github.com/u-root/u-root/cmds/uniq", "github.com/u-root/u-root/cmds/unshare", "github.com/u-root/u-root/cmds/validate", "github.com/u-root/u-root/cmds/vboot", "github.com/u-root/u-root/cmds/wc", "github.com/u-root/u-root/cmds/wget", "github.com/u-root/u-root/cmds/which", "github.com/u-root/u-root/cmds/wifi", "github.com/u-root/u-root/cmds/xinit", "github.com/erroneousboat/slack-term", "github.com/elves/elvish", "github.com/SvenDowideit/u-root/_examples/uinit"}, TempDir:"/tmp/u-root068099761/builder826207324"}: error formating: format.Node internal error (1:14: expected ';', found '-')
Makefile:8: recipe for target 'build' failed
make: *** [build] Error 1
```

vs when I fork and rename:

```
u-root \
        -format=cpio \
        -build=bb \
        -o initramfs.cpio \
        ./cmds/* \
        github.com/elves/elvish \
        github.com/SvenDowideit/slackterm \
        github.com/SvenDowideit/u-root/_examples/uinit
2018/01/16 10:32:41 Disabling CGO for u-root...
2018/01/16 10:32:41 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/OSM/HOME-MEL/dow184/go CGO_ENABLED=0
2018/01/16 10:33:04 Filename is initramfs.cpio
```

hugelgupf, 2018-01-16:
Huh. Cool! Looking into it.

hugelgupf, 2018-01-16:
Argh. I see. We use the directory name as the command's package name. But obviously the set of characters allowed in go package names is smaller than the possible names for Linux directories.

hugelgupf, 2018-01-16:
You should be able to build this in source mode or binary mode in the meantime.

hugelgupf, 2018-02-12:
Adding "init" to the list: 

```
./cmd_init.go:3:8: cannot import package as init - init must be a func
```

hugelgupf, 2020-09-24:
https://github.com/u-root/gobusybox/issues/41 will solve this

closed 2020-09-24 by hugelgupf