u-root/u-root #569: How to add init scripts or extend init process?

pwFoo, 2018-01-23:
Is it possible to extend u-root init process or add something like init.d scripts?
Haven't found documentation about.

rminnich, 2018-01-23:
we've had this discussion on slack.

This is the thinking so far:
- we don't want systemd
- we don't want init.d scripts
- we don't want upstart
- "scripts" need to be go programs so we can at least verify they don't contain dumb errors
- "scripts" also should be able to dynamically compile like any other u-root command if desired
- "scripts" should not be shell scripts, since we have no agreement on a shell yet

-services are written in Go, and are controllable via some sort of REST or gRPC interface
- it's 2018, and we don't have to restrict to stdin/stdout/signal to control things
- the service that manages the services is itself a service that follows these rules

we should consider systems like https://github.com/hashicorp/go-plugin

rminnich, 2018-01-23:
Oh and thanks for making this issue.

pwFoo, 2018-01-24:
@rminnich 
At the moment I play around with a custom (shell) init script. Maybe I can convert it to go (mount cgroups, prepare some things, ...) in the future.

init.d was just a known example. I dont won't to use systemd, upstart, sysvinit, ...
That's why I play around with custom init, busybox, linuxkit and u-root ;)

I just need a way to execute a script (I hope it could be go soon...) after system was bootet. :)

pwFoo, 2018-01-25:
@rminnich
Is it possible to execute a (Go) script after boot finished instead of just start a shell?

hugelgupf, 2018-01-25:
Yes. Include any Go command named uinit.

pwFoo, 2018-01-25:
Thanks! I'll convert my shell script to go and try it :)

bensallen, 2018-02-03:
Also see #60 to track the work around what @rminnich describes.

rjoleary, 2018-07-15:
The two relevant features here are:

* uinit (user init): any command named `uinit` will execute on startup
* SoS: Service of Services for daemons such as wifi

The rest is a duplicate of #60

rjoleary, 2018-07-15:
this is a duplicate of #60

closed 2018-07-15 by rjoleary

pwFoo, 2018-07-15:
@rjoleary
Is SoS documentation / example available?

rjoleary, 2018-07-15:
Documentation: https://godoc.org/github.com/u-root/u-root/pkg/sos (The documentation is limited at the moment; hopefully, this can be improved. Just filed #809. ) SoS is not about init scripts, rather it manages services.

Examples:

* wifi: https://github.com/u-root/u-root/blob/master/cmds/wifi/server.go
* upspin: https://github.com/u-root/u-root/blob/master/cmds/upspin_sos/server.go

@jbeta51 and @TrevorFarrelly are working on another service -- they know a lot more about SoS.