u-root/u-root #572: Add directories with -files

pwFoo, 2018-01-24:
Hi,

to add needed modules (#525) or directory of app files (recursive) it would be useful to add directories with `-files`. At the moment I use a find command to get all the files inside of my prepared `modules` directory to add it recursive.

How to add a directory recursive to the u-root generated initrd?

pwFoo, 2018-01-24:
I tried a find command to add all the files inside of my prepared `modules` directory recursive, but files and directories are missing inside of the initrd.

hugelgupf, 2018-01-24:
-files should already support directories - have you tried that? If that
fails for some reason, that'd be a bug.

pwFoo, 2018-01-24:
I can't add directories.
```
u-root -files "/lib/modules:lib/modules" -files "/usr/bin/runc:usr/bin/runc" -build=bb -o /initrd ./cmds/*
2018/01/24 16:44:48 Disabling CGO for u-root...
2018/01/24 16:44:48 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/go CGO_ENABLED=0
2018/01/24 16:44:48 Filename is /initrd
```

/lib/modules size
```
du -sh /lib/modules/
174.1M  /lib/modules/
```

```
-rwxr-xr-x    1 root     root       19.0M Jan 24 16:45 /initrd
```

Test with qemu...

Messages in the end of the boot process
```
[...]
2018/01/24 16:55:09 Created dir "/buildbin" (mode 0777)
2018/01/24 16:55:09 Created dir "/ubin" (mode 0777)
2018/01/24 16:55:09 Created dir "/tmp" (mode 0777)
2018/01/24 16:55:09 Created dir "/env" (mode 0777)
2018/01/24 16:55:09 Created dir "/tcz" (mode 0777)
2018/01/24 16:55:09 Created dir "/lib" (mode 0777)
2018/01/24 16:55:09 Created dir "/usr/lib" (mode 0777)
2018/01/24 16:55:09 Created dir "/go/pkg/linux_amd64" (mode 0777)
2018/01/24 16:55:09 Created dir "/etc" (mode 0777)
2018/01/24 16:55:09 Created dir "/proc" (mode 0555)
2018/01/24 16:55:09 Created mount -t "proc" -o  "" "/proc" flags 0x0
2018/01/24 16:55:09 Created mount -t "tmpfs" -o  "" "/tmp" flags 0x0
2018/01/24 16:55:09 Created dir "/dev" (mode 0777)
2018/01/24 16:55:09 Created dev "/dev/tty" (mode 020666; magic 1280)
2018/01/24 16:55:09 Created dev "/dev/urandom" (mode 020444; magic 265)
2018/01/24 16:55:09 Created dev "/dev/port" (mode 020640; magic 260)
2018/01/24 16:55:09 Created mount -t "devtmpfs" -o  "" "/dev" flags 0x0
2018/01/24 16:55:09 Created dir "/dev/pts" (mode 0777)
2018/01/24 16:55:09 Created mount -t "devpts" -o newinstance,ptmxmode=666,gid=5,mode=620 "" "/dev/pts" flags 0x0
2018/01/24 16:55:09 Created symlink "/dev/ptmx" -> "/dev/pts/ptmx"
2018/01/24 16:55:09 Created dir "/dev/shm" (mode 0777)
2018/01/24 16:55:09 Created mount -t "tmpfs" -o  "tmpfs" "/dev/shm" flags 0x0
2018/01/24 16:55:09 Created dir "/sys" (mode 0555)
2018/01/24 16:55:10 Created mount -t "sysfs" -o  "sys" "/sys" flags 0x0
2018/01/24 16:55:10 Created mount -t "tmpfs" -o  "cgroup" "/sys/fs/cgroup" flags 0x0
2018/01/24 16:55:10 Created dir "/sys/fs/cgroup/memory" (mode 0555)
2018/01/24 16:55:10 Created dir "/sys/fs/cgroup/freezer" (mode 0555)
2018/01/24 16:55:10 Created dir "/sys/fs/cgroup/devices" (mode 0555)
2018/01/24 16:55:10 Created dir "/sys/fs/cgroup/cpu,cpuacct" (mode 0555)
2018/01/24 16:55:10 Created symlink "/sys/fs/cgroup/cpu" -> "/sys/fs/cgroup/cpu,cpuacct"
2018/01/24 16:55:10 Created symlink "/sys/fs/cgroup/cpuacct" -> "/sys/fs/cgroup/cpu,cpuacct"
2018/01/24 16:55:10 Created mount -t "cgroup" -o memory "cgroup" "/sys/fs/cgroup/memory" flags 0x0
2018/01/24 16:55:10 Created mount -t "cgroup" -o freezer "cgroup" "/sys/fs/cgroup/freezer" flags 0x0
2018/01/24 16:55:10 Created mount -t "cgroup" -o devices "cgroup" "/sys/fs/cgroup/devices" flags 0x0
2018/01/24 16:55:10 Created mount -t "cgroup" -o cpu,cpuacct "cgroup" "/sys/fs/cgroup/cpu,cpuacct" flags 0x0
2018/01/24 16:55:10 Run &{/inito [/inito] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42007d9a0 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
2018/01/24 16:55:10 fork/exec /inito: no such file or directory
2018/01/24 16:55:10 Run &{/bbin/uinit [/bbin/uinit] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42007da40 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
2018/01/24 16:55:10 fork/exec /bbin/uinit: no such file or directory
2018/01/24 16:55:10 Run &{/bbin/rush [/bbin/rush] []  0xc42000c010 0xc42000c018 0xc42000c020 [] 0xc42007dae0 <nil> <nil> <nil> <nil> false [] [] [] [] <nil> <nil>}
% [    4.651084] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input3
```

Directory doesn't work
```
% ls -l /lib/modules
2018/01/24 16:56:33 /lib/modules: lstat /lib/modules: no such file or directory
```

File works
```
ls -l /usr/bin/runc
-rwxr-xr-x 0 0 8056456 Jan  1 00:00 runc
```

insomniacslk, 2018-01-24:
you should pass `-files` only once, separating by spaces each file/dir you want to include. With the command line you're using, the last occurrence of `-files` overrides the previous one(s). E.g. in your case you should invoke:

```
u-root -files "/lib/modules:lib/modules /usr/bin/runc:usr/bin/runc" -build=bb -o /initrd ./cmds/*
```

pwFoo, 2018-01-24:
Thanks, @insomniacslk !
Works fine with correct syntax.

hugelgupf, 2018-01-25:
Huh. We should allow both those cases, imho, and it shouldn't be too hard with Go's flags. I'll just leave this open to remember to do that.

insomniacslk, 2018-01-25:
According to https://golang.org/src/flag/example_test.go#L33 this can be done quite easily. I've given it a try and made https://github.com/u-root/u-root/pull/574

bensallen, 2018-02-01:
Closing with #574 merged.

closed 2018-02-01 by bensallen