u-root/u-root #576: If cmd have to be compiled from source runtime inform user about?

pwFoo, 2018-01-26:
After built u-root initrd with `-build=source` it tooks a long time to but and seems to hang without an error message. Some time (minutes? used qemu without kvm...) later it reached the shell.

I think the commands have to be compiled from source during boot and it looks like it got stuck?

Could you echo an information the the shell before the command start to compile from source? Just to inform the user about? Would be nice to know during boot process or delayed command execution.

insomniacslk, 2018-01-26:
What you're seeing matches what happens when you are using full sw emulation and source builds, so yes, that's probably your case.
This issue shows up only in this case (slow emulation) so I would be wary of printing anything when a command is first compiled. Maybe a better approach is to print out a message, before `rush` starts, saying something like "hey this is a source build, now we are compiling rush/whatever command, please wait". That would work well either if it's slow or fast.
That's my opinion of course, the project maintainers will be able to give you a more authoritative answer

pwFoo, 2018-08-24:
@insomniacslk 
Is such an information added to source build boot process? Issue could be closed then

closed 2018-10-28 by pwFoo