u-root/u-root #579: dhclient problems

pwFoo, 2018-01-27:
Created an initrd with `-build=source` and tried to get internet connection working with `dhclient` without success.
Maybe related with #500 and #448.

Loaded NIC driver with modprobe and get ip address with `dhclient -ipv4`.

Without `-verbose` it looks like dhcp fails, but verbose outpute show me a successful IPv4 lease (netmask, router, routerName, ip added to interface).

Then dhclient try to renew a not existing DHCPv6 lease until timeout. Shouldn't dhclient end after get an IPv4 lease? I have to interrupt dhclient to stop it "resend DHCPv6 Solicit Message".

`ip a` show a valid ip address and `ip r` two routes, but not in a readable output. Gateway is `01C8A8C0`?
Ping fails with `Read failed` and `i/o timeout`.

pwFoo, 2018-01-27:
Also can't ping the default gateway.

insomniacslk, 2018-01-27:
This is probably due to the defaults for `-retry` and `-renewals`:

```
  -renewals int
    	Number of DHCP renewals before exiting. -1 means infinity (default -1)
  -retry int
    	Max number of attempts for DHCP clients to send requests. -1 means infinity (default -1)
```

Does it work as you expect if you use `-retry 1 -renewals 0`? I was waiting on https://github.com/u-root/u-root/pull/517 before starting working on the `dhclient` behaviour and defaults.

@hugelgupf should we change these defaults, or, probably better, making `dhclient` go to background by default after a successful lease?
This would also mean adding an option to run in foreground.

pwFoo, 2018-01-28:
Go to background sounds like a daemon?

Default it should endlich after successful lease I think.

pwFoo, 2018-01-28:
But network connection doesn't work. Is there a problem with routes or something else?

insomniacslk, 2018-01-28:
yes, it is a daemon. Normally ISC's `dhclient` runs in foreground until an interface is configured, then if goes to background to handle renewals, disconnections etc.

Not sure what exactly the problem you're seeing is, but it could be related to https://github.com/u-root/u-root/issues/500 as you suggest. I haven't tried dhclient in a v4 environment yet so I don't have an answer off top of my head. However I would run dhclient with `-verbose -retry 1 -renewals 0`, check the output of `ip a`, the content of `/proc/net/route` (and /proc/net/ipv6_route` too if you're in a dual-stacked network). Can you do that and paste the output here?

hugelgupf, 2018-02-10:
I'm going to park this issue with me for now, as I'm doing a whole bunch of dhclient work. I may not actually get to this, though.

pwFoo, 2018-02-17:
Tested with latest u-root and it works fine.
```
dhclient -ipv6=false -retry 1 -renewals 0 
```

insomniacslk, 2018-02-17:
Cool :) we also have plans to make it fork in background by default, see https://github.com/u-root/u-root/issues/632

pwFoo, 2018-05-05:
With latest u-root dhclient doesn't work for me?
`dhclient eth0`
`dhclient -verbose -retry 1 -renewals 0 eth0`

Before updated u-root `dhclient eth0` also without options worked fine. I think I worked some weeks (3 or 4?) with a local version.

bluecmd, 2018-11-04:
What's left on this bug that is not tracked by #632 ?

pwFoo, 2018-11-04:
Dhclient works, but only foreground. I start it, wait for a lease and then kill dhclient. A --background option would bei nice

bluecmd, 2018-11-04:
Right, and I think that's what #632 is tracking, and it has less discussions - so I'll close this issue in favor for that one.

closed 2018-11-04 by bluecmd