u-root/u-root #580: pxeboot: gpgv as a first-class concept.

hugelgupf, 2018-01-29:
gpg verify signatures for kernel and initrd, as well as cmdline string. pxeboot should have a policy flag that let's you decide whether to boot, but emit a warning, or not to boot when verification fails.

This could be done in one of several ways: 

 * Assume that whatever the signature is contained in "fileURI.sig" on the TFTP/http/whatever server.
 * Add config keys to our pxelinux config parser for signature files.

rjoleary, 2018-01-29:
Another useful policy would be: On verification failure, retry with appropriate backoff.

It would probably be good to put all the security considerations in one document and have it reviewed by someone good with crypto.

* How are kernel, initrd and cmdline combined to be passed to gpgv?
* Can trusted keys be revoked?
* Should pxeboot also guard against downgrades?

lzrd, 2018-01-29:
If there is a hardware root-of-trust on the system, and the policy is to
always boot, then the RoT needs to be told about the success/failure of the
signature check. There should at least be a TODO() on that point.
One could also try a different boot path until a valid image is found, i.e.
cases can be made for a variety of policies and the boot mechanisms should
be able to accommodate a variety of policies.

hugelgupf, 2018-01-29:
cc @jonmccune

hugelgupf, 2018-02-10:
Just as an FYI for people here, I have a proposal forthcoming. I think the doc needs some touch up and security review before I want to release it to the public, though :)

jonmccune, 2018-02-11:
Ack.

-Jon