u-root/u-root #581: u-root growth: different purposes

hugelgupf, 2018-01-29:
Here's another conversation I would like to start. u-root is growing for different projects with different use cases at this point. We're gaining bootloader features (cmds/pxeboot, cmds/boot), as well as possibly two different init systems (uinit for firmware init, other init for NiChrome init).

One of the issues with this is that the *default* u-root initramfs will keep growing in size and complexity. NERF builds will include random NiChrome tools, like WiFi-related stuff. NiChrome builds will include random NERF tools, like pxeboot and boot.

I think we need to figure out how to separate the different purposes from the base build of u-root. The vague repos I have in my head:

 * base u-root (u-root/u-root repo): basic busybox-like utilities / GNU coreutils-like utilities.
 * NERF u-root (u-root/NERF): pxeboot, disk boot tools, etc.
 * NiChrome u-root (u-root/NiChrome): WiFi tools, the fancier init system, etc.

The idea would be that any shared code could be in the main u-root repo as packages, but any cmds covering the NiChrome or NERF use case specifically should be in a separate repo.

cc @ganshun @rminnich @thienhoang23

rminnich, 2018-01-30:
we need to plan for personalities I guess, and provide a default personality, and let users override it?

pwFoo, 2018-02-01:
Maybe a custom builds need cmds of both? So all the features should be available. What so you think about pre-defined cmd lists or cmd "groups" to use?

hugelgupf, 2018-02-01:
I'm thinking that the default set of commands should be the lowest common denominator - the original u-root with just the busybox-like tools.

Some NiChrome things are already in their own repo, (e.g. install), whereas stuff like the wifi command is in u-root.

All NERF-specific commands are currently in the u-root repo. I'm thinking that we may want to create a repo like u-root/boot that contains all the booting tools. 

In the other hand, this somewhat hinders discoverability of features we've implemented. We'd need (well, we still need) better documentation of what's where and how it can be used.

insomniacslk, 2018-02-01:
I entirely agree on the minimal subset and the additional repos to add feature sets instead of single commands/features, and we would benefit from this approach too. It would be also good to keep `init` as a candidate to become an easily replaceable component for all the build modes. We already discussed this on Slack and I know this is coming, just keeping track of it here too :)

pwFoo, 2018-05-31:
u-root is awesome, but it lacks documentation of many features? How to use packages like wifi and much more?

rjoleary, 2021-06-07:
I feel this has been mostly solved by the sub-directories under cmds/ as well as the templates file https://github.com/u-root/u-root/blob/master/templates.go.

closed 2021-09-20 by rjoleary