u-root/u-root #612: ps_test: ps should be able to absorb disappearing processes

hugelgupf, 2018-02-11:
https://circleci.com/gh/u-root/u-root/3174?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

```
--- FAIL: TestPsExecution (0.01s)
	ps_test.go:18: Loading Table fails on some point; read /proc/5272/stat: no such process
FAIL
FAIL	github.com/u-root/u-root/cmds/ps	0.090s
```

Processes exit as they are wont to do, and checking procfs is racy at all levels, obviously. `ps` should be able to deal with that.

hugelgupf, 2018-02-11:
cc @rminnich @thienhoang23 thien could look at this if you think it'd be a good bug?

bensallen, 2018-02-13:
ps deals with this by ignoring the error: https://github.com/u-root/u-root/blob/master/cmds/ps/ps.go#L227. ps_test.go does not. Maybe we should just mockup a number of stat files for this test?

I suspect enumerating and reading /proc/<pid>/stat can be done in a tighter loop, feeding another goroutine that does the processing of the data. This is just racing the condition faster, however. Maybe one could use inotify on /proc to get the kernel to push changes from the time the readdir is done to the time all the reads of stat files are completed? Check the accumulated errors against directories that where removed, before calling it a real error.

GanShun, 2018-08-09:
I ran into this again today:

https://circleci.com/gh/u-root/u-root/13543?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

@hugelgupf could you rerun #871 for me when you can? I don't have write permissions or something and it's not allowing me to re run the tests.

hugelgupf, 2019-09-23:
This is the same as #511.

closed 2019-09-23 by hugelgupf