u-root/u-root #630: PXE: boot with empty configuration

vejmarie, 2018-02-14:
ends up with a traceback. (I know this is definitly a stupid thing as there is no reason to try to boot if the tftp server is not configured, but we shouldn't crash)

% pxeboot
2018/02/14 15:43:44 Attempting to get DHCP lease on eth0
2018/02/14 15:43:44 Got lease on eth0
2018/02/14 15:43:44 Got configuration: &{<nil> <nil> }
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x461bb2]

goroutine 1 [running]:
io.copyBuffer(0x918960, 0xc42000c098, 0x0, 0x0, 0xc420126000, 0x8000, 0x8000, 0xc4200f79a0, 0xc4200f79a8, 0x15)
	/usr/local/go/src/io/io.go:392 +0x102
io.Copy(0x918960, 0xc42000c098, 0x0, 0x0, 0xc42000c098, 0x0, 0x0)
	/usr/local/go/src/io/io.go:362 +0x68
github.com/u-root/u-root/bbsh/cmds/pxeboot.copyToFile(0x0, 0x0, 0x0, 0x0, 0x0)
	/home/ubuntu/gopath/src/github.com/u-root/u-root/bbsh/cmds/pxeboot/pxeboot.go:32 +0xd7
github.com/u-root/u-root/bbsh/cmds/pxeboot.Netboot(0x0, 0x0)
	/home/ubuntu/gopath/src/github.com/u-root/u-root/bbsh/cmds/pxeboot/pxeboot.go:113 +0x716
github.com/u-root/u-root/bbsh/cmds/pxeboot.Main()
	/home/ubuntu/gopath/src/github.com/u-root/u-root/bbsh/cmds/pxeboot/pxeboot.go:150 +0xa2
main._forkbuiltin_pxeboot(0xc420090620, 0xc42006a2c0, 0x7fffef906ef6)
	/home/ubuntu/gopath/src/github.com/u-root/u-root/bbsh/cmd_pxeboot.go:6 +0x20
main.main()
	/home/ubuntu/gopath/src/github.com/u-root/u-root/bbsh/rush.go:207 +0x356
wait: exit status 2

rminnich, 2018-02-14:
oh, nice. Reminds me of the days when people would do this to exit a program
if (something) x = 1/0;

yeah, divide by zero trap. :-)

vejmarie, 2018-02-14:
which was still better than the stupid mode that SGI put in place a while ago where a 0 divide was not generating traps as to allow some scientist to run there code without debugging them :( (especially in astronomy)

closed 2024-02-20 by hugelgupf