u-root/u-root #640: elvish in bb mode: Cannot connect to daemon

pwFoo, 2018-02-17:
elvish daemon / history doesn't work in bb mode
```
% elvish
no scripts/args yet

Cannot connect to daemon: daemon unreachable after waiting for 1s
Daemon-related functions will likely not work.
Failed to initialize command history; disabled.
~/> 
```

Works fine in source mode

rminnich, 2018-02-17:
yes. I'm ok with that. Elvish is insane. It opens up a port and lets random processes connect to it and execute commands. And the daemon is part of that. 

If you want to fix elvish by removing the daemon I'm fine with that. But overall I think it has a lot of bad ideas in it and I at least no longer want it in this project.

Let's try to find a better shell. Anyone interested in porting pdksh to Go?

insomniacslk, 2018-02-17:
We should probably also remove the advice of using `elvish` from https://github.com/u-root/u-root/blob/master/README.md .

However until we have a new shell, a workaround is to just pass `-files /bin/bash` when building the ramfs. This will significantly increase the ramfs size and it's against the philosophy of this project, but still worth remembering that it's an option

pwFoo, 2018-02-17:
Searched for go shell and found https://github.com/michaelmacinnis/oh. Tested successful in source mode. Binary is 3,5MB
```
-rwxr-xr-x    1 root     root        3.5M Feb 17 22:11 oh
```

Why with rush and oh inside of u-root auto complete doesn't work? With "oh" shell outside of u-root auto complete works for directories if I start with "/" or "./". Outside of u.root rush doesn't start ([ERRO] STDIN not detected) to test it too. 
Bootet into u-root tab to autocomplete doesn't work for me.

hugelgupf, 2018-02-17:
It's still a bug in bb - we should make this work anyway to further the claim of "bb mode can support any reasonable go program"

pwFoo, 2018-10-28:
@hugelgupf
Current u-root works fine with elvish shell.
Fixed? Should we close the issue?

closed 2018-10-29 by rjoleary