u-root/u-root #642: ping bug? read failed [...] i/o timeout

pwFoo, 2018-02-18:
Maybe a bug in ping package?

Internet connection works fine (tested with `wget http://google.com` and downloaded an index.html file with content), but ping fails with a 

> read failed [...] i/o timeout

ipv4 network connection (nic module loaded, dhclient to get an valid ipv4 address).

insomniacslk, 2018-02-18:
I believe @claudiozz and/or @tfg13 were working on fixing it

tfg13, 2018-02-18:
Yup, we got the same error message, but on v6 (network was also working fine otherwise) @claudiozz did some initial investigation, and IIRC it pointed at something in the v6 codepath. So not sure if this is the same thing.

That being said, we will probably fix it at some point, but there is no ETA since this is not critical for us at this time.

pwFoo, 2018-02-18:
OK, thanks. It's a low prio issue ;)

pwFoo, 2018-05-20:
I created a initrd with alpine linux base with u-root additions and ping binary /bbin/ping works fine.

Is it solved or is the alpine rootfs the workaround?

pwFoo, 2018-10-28:
Works fine in current u-root.

closed 2018-10-28 by pwFoo